mainsail-crew/moonraker-timelapse #8: still having some config issues

jbrandek, 2021-10-31:
some fixed but still having this ones:

check the logfile and fix the issue.

Following plugin has an error:

timelapse
Invalid config option 'output_framerate: 20 # Note: this will be ignored if variable_fps is enabled' detected in section [timelapse]. To solve this issue check your moonraker config. This option is invalid and has to be deleted or fixed. In the future this will result in a startup error.

Invalid config option 'time_format_code: %Y%m%d_%H%M' detected in section [timelapse]. To solve this issue check your moonraker config. This option is invalid and has to be deleted or fixed. In the future this will result in a startup error.

Invalid config option 'snapshoturl: http://localhost:8080/?action=snapshot' detected in section [timelapse]. To solve this issue check your moonraker config. This option is invalid and has to be deleted or fixed. In the future this will result in a startup error.

Invalid config option 'pixelformat: yuv420p' detected in section [timelapse]. To solve this issue check your moonraker config. This option is invalid and has to be deleted or fixed. In the future this will result in a startup error.

Invalid config option 'previewimage: True' detected in section [timelapse]. To solve this issue check your moonraker config. This option is invalid and has to be deleted or fixed. In the future this will result in a startup error.

jbrandek, 2021-10-31:
Sorry my bad learned the hard way you cannot comment in same line as a value is set

now the only thing is not working is previewImage

Invalid config option 'previewimage: True' detected in section [timelapse]. To solve this issue check your moonraker config. This option is invalid and has to be deleted or fixed. In the future this will result in a startup error.

FrYakaTKoP, 2021-10-31:
@jbrandek could you please provide a moonraker.log?

FrYakaTKoP, 2021-10-31:
it seems like moonraker config helper doesn't like the uppercase "I" in the config name. 

i pushed fa81ef3 to fix it.

thanks for finding and reporting it, since i use a alpha version of mainsail which includes those settings in the Ui i would never catch those!

jbrandek, 2021-10-31:
Thanks for making the plugin and fix and reply so fast

FrYakaTKoP, 2021-10-31:
your welcome again thanks for testing and feedback

closed 2021-10-31 by FrYakaTKoP