mainsail-crew/moonraker-timelapse #10: Unknown command: "PAUSE_BASE"

falo2k, 2021-11-05:
Just started testing out the plugin and it's working well so far.  Cheers for all your work on this.  I have run into one error about an undefined "PAUSE_BASE" in my console.  You use it twice in the macros - it assumes that people have renamed stock PAUSE to PAUSE_BASE rather than something else (I use BASE_PAUSE, for example).

zellneralex, 2021-11-06:
Hi yes, there is no other way as to assume a name. We used the latest recommendation of fluidd/mainsail.

These names is selected so that if you type HELP the 2 macros are listed next to each other.

We will make a note in the documentation to make that clear, but that’s all we can do as there is no other way at the moment. klipper does not let us detect how the klipper original PAUSE macro is renamed and we need that for the continuance feature to get the toolhead parked for the foto without the need to tweak the delay.

falo2k, 2021-11-06:
Yeah, completely understand the challenge.  Documenting it seems like the way forward.

FrYakaTKoP, 2021-12-05:
PR  #20 will fix this no more documentation needed

FrYakaTKoP, 2021-12-05:
Pr #20  is merged which resolves this issue

closed 2021-12-05 by FrYakaTKoP