mainsail-crew/moonraker-timelapse #12: FluiddPi cannot load the plugin

phoenixfire425, 2021-11-15:
This is what the UI says
fluidd warnings found.
Moonraker has failed plugins, please check your logs, update your configuration and restart moonraker.
timelapse
Moonraker plugin configuration can be found here.

Here is the error from the log

2021-11-15 20:24:40,376 [moonraker.py:load_component()] - Unable to load component: (timelapse)
Traceback (most recent call last):
  File "/home/pi/moonraker/moonraker/moonraker.py", line 204, in load_component
    component = load_func(config)
  File "/home/pi/moonraker/moonraker/components/timelapse.py", line 677, in load_component
    return Timelapse(config)
  File "/home/pi/moonraker/moonraker/components/timelapse.py", line 131, in __init__
    full_access=True
TypeError: register_directory() got an unexpected keyword argument 'full_access
[moonraker (3).log](https://github.com/mainsail-crew/moonraker-timelapse/files/7541460/moonraker.3.log)
'

FrYakaTKoP, 2021-11-19:
@phoenixfire425 your moonraker is to old the argument "full_access" was added recently (23 days ago), just update moonraker and it should work.

commit https://github.com/Arksine/moonraker/commit/734f295822a733958275ff2a13c2cb9e8f75c3e6 added the feature it's missing.

FrYakaTKoP, 2021-11-21:
no further reaction of OP and problem should be resolved, closing this issue

closed 2021-11-21 by FrYakaTKoP