mainsail-crew/moonraker-timelapse #13: Cannot install timelapse with 2 instances of klipper & mainsail

jpbnto, 2021-11-17:
~$ cd ~/
git clone https://github.com/mainsail-crew/moonraker-timelapse.git
bash ~/moonraker-timelapse/install.sh

~ $ Cloning into 'moonraker-timelapse'...
remote: Enumerating objects: 258, done.
remote: Counting objects: 100% (258/258), done.
remote: Compressing objects: 100% (181/181), done.
remote: Total 258 (delta 127), reused 128 (delta 50), pack-reused 0
Receiving objects: 100% (258/258), 132.17 KiB | 1.41 MiB/s, done.
Resolving deltas: 100% (127/127), done.
Klipper service not found, please install Klipper first

FrYakaTKoP, 2021-11-19:
@jpbnto 
As i already answered in the mainsail discord: I just forget about multi instance. We will work on a solution but right now timelapse isn't multi instance compatible.

What would need to change:
- the dummy service (so it restarts only the correct instance, etc.)
- the installer to detect multiple instances and handle it correctly
- the frame path would need to be separated so each instance has there own folder (this is already possible since the path is configurable in moonraker.conf)

What you can do at the moment. You could create the symlinks of the timelapse.cfg and timelapse.py manually. Then without a working dummy service you would need to restart moonraker and klipper of each instance by yourself.

manually linking of the files would be something like this
(warning this isn't copy pasta ready! you need to change your paths by yourself!)
```
cd ~/moonraker-timelapse/
ln -sf "./component/timelapse.py" "{moonraker instance path}/moonraker/components/timelapse.py"
ln -sf "./klipper_macro/timelapse.cfg" "{klipper instance config path}/timelapse.cfg"
```

ronnievv82, 2022-03-23:
So what exactly do I need to do to get Timelapse to see my Klipper install. I've made the sym links and edited the frame_path, but am still not showing in Mainsail? Thank you.

harellevy, 2022-05-07:
I got "Klipper service not found, please install Klipper first" here also. Not sure why I can't make it work. tried to copy to printer1 configuration as mentioned but couldn't make this work. Please help.

FrYakaTKoP, 2022-05-14:
@harellevy  the installer will not work, but it doesn't has to! If you linked the plugin and macro file you don't need the installer to run

Fuckingnameless, 2022-05-18:
> you also need to change the frame_path: config in moonraker.conf so each instance has it's own dedicated directory

where is this?

Fuckingnameless, 2022-05-18:
ls: cannot access '/home/pi/klipper_config/webcam*.txt': No such file or directory
where is the file that´s pointing at that?

FrYakaTKoP, 2022-05-18:
@Fuckingnameless  the webcam.txt has nothing todo with timelapse. Only thing is you need to change the resolution of the stream inside the webcam.txt which then changes the resolution of the timelapse. But the timelapse plugin doesn't interact with the webcam.txt at all. This is a OS (MainsailOs/Fluiddpi or Kiauh) related question, please open a issue there.

Eg0blast, 2022-10-02:
@FrYakaTKoP thank you very much, your copy comment above worked perfectly.

zellneralex, 2023-02-14:
Multi instance install added with 3ac026158b2b849456bb9bf20c7663c667fa4d84

closed 2023-02-14 by zellneralex