mainsail-crew/moonraker-timelapse #15: wont take next frame

drveni, 2021-11-21:
hi

i use fluidd, and time-frame/lapse
on my command from terminal "TIMELAPSE_TAKE_FRAME" he take picture on my saved directory /tmp/timelapse/

and it names frame000001.jpg

on next command TIMELAPSE_TAKE_FRAME he doesn't take next number frame00002.jpg, he just overite 00001.jpg... what im doing wrong?

when im download pic from raspi (cyberduck) to mac.. i'm only can open in photoshop, in normal pic view program got some error...

sorry for bad eng

WickedInvi, 2021-11-21:
I have the same issue. Also, the Size of the frame is 0 KB
[image]

FrYakaTKoP, 2021-11-21:
@WickedInvi @drveni 
my guess is you entered the webcamstream to snapshoturl instead of the snapshot. You should see a take frame timeout in moonraker.log.

The snapshot url should end like ``?action=snapshot``. Please provide moonraker.log to do further analysis if your snapshoturl is correct (check it with the browser, it should return a single jpg frame)

drveni, 2021-11-21:
> @WickedInvi @drveni my guess is you entered the webcamstream to snapshoturl instead of the snapshot. You should see a take frame timeout in moonraker.log.
> 
> The snapshot url should end like `?action=snapshot`. Please provide moonraker.log to do further analysis if your snapshoturl is correct (check it with the browser, it should return a single jpg frame)
> 
> edit: We could track down WickedInvi's issue on the klipper discord: His mjpg streamer required authentication for the snapshot, because he had reinstalled the streamer with a guide which added authentication.

sorry for my mistake I thought it was web. whit snapshot works great!! sorry sorry !!!

and second

18:10:34 
// Unknown command:"PAUSE_BASE"
18:10:34 
Done printing file
18:10:35 
// Timelapse: Rendering finished
18:10:35 
// Unknown command:"RESUME_BASE"

I see that there is already an open topic whit same error but im send here my log
[moonraker.log](https://github.com/mainsail-crew/moonraker-timelapse/files/7576694/moonraker.log)

FrYakaTKoP, 2021-11-21:
@drveni 

your Pause and resume macros were old, in newer installations those macros include ``rename_existing: PAUSE_BASE`` for pause and ``rename_existing: RESUME_BASE`` for resume. Either replace all older calls (normally BASE_PAUSE & BASE_RESUME) or update your macros with this here:

https://github.com/mainsail-crew/MainsailOS/blob/master/src/modules/mainsail/filesystem/home/pi/klipper_config/mainsail.cfg#L33

I recommend to change this because it's now a quasi standard most images adapted and if not changed you can't use the parkhead feature. Since you posted the output of the render macro this isn't a real issue. Because in the render macro, the pause is only used to block further gcode execution which is only needed if a user like to shutdown the pi automatically afterwards render is finished. Most people also don't need the render macro anyway, since mainsail v2.1 beta does include a render button with some render related settings.

Since the original question/issue is answered, please close this issue and open up a new one if you have further questions or consider joining the klipper (or mainsail) discord to get direct support

drveni, 2021-11-21:
> @drveni
>
> your Pause and resume macros were old, in newer installations those macros include `rename_existing: PAUSE_BASE` for pause and `rename_existing: RESUME_BASE` for resume. Either replace all [...]

oh, thank you so much for your help. i have been using this for a long time and haven't seen a problem. but I will definitely remake this !!!

thanks again for help!

FrYakaTKoP, 2021-11-21:
your welcome!

closing this issue since original problem is solved

closed 2021-11-21 by FrYakaTKoP