mainsail-crew/moonraker-timelapse #16: Add a delayed last frame

Sekenenz, 2021-11-21:
Is it possible to modify the timelapse macro to get the last frame after the execution of the end g-code (to not have the hot-end in the last frame)? Or I missed some settings?

Btw I'm not using (and prefer to not use) the park function to avoid stringing and prolonged times.

FrYakaTKoP, 2021-11-22:
@Sekenenz 
I never thought of such feature, but actually you should be able to use the existing macro with parameters in your endgcode to achive such.

add this to your end gcode
``` 
_SET_TIMELAPSE_SETUP PARK_ENABLE=True
TIMELAPSE_TAKE_FRAME
_SET_TIMELAPSE_SETUP PARK_ENABLE=False
```
The parkposition, speed and so on can be changed within mainsail just enable the park feature temporarily to change the setting to your liking. Then deactivate it again, if you don't like to use it while printing.

Also change the duplicatelastframe setting, to last this special frame a couple seconds

Sekenenz, 2021-11-22:
I will asap thanks.

Ps. The printhead is already set to move in the corner in my end-gcode, so I can just add TIMELAPSE_TAKE_FRAME to it and omit the park, right?

FrYakaTKoP, 2021-11-22:
yes, you can do that

Sekenenz, 2021-11-22:
Just tested and unfortunately it doesn't work:

Adding only
`TIMELAPSE_TAKE_FRAME`
doesn't work.

Adding
```
_SET_TIMELAPSE_SETUP PARK_ENABLE=True
TIMELAPSE_TAKE_FRAME
_SET_TIMELAPSE_SETUP PARK_ENABLE=False
```
It works but the printer enter a loop and I have to emergency stop to block it.

The parkpos is set to back_left and duplicatelastframe to 10 (but it's always ignored)

FrYakaTKoP, 2021-11-22:
@sekenenz
could you please provide your moonraker.log?

Sekenenz, 2021-11-22:
https://drive.google.com/file/d/1mp-lRdDJsTrQaXr_b8ebFWIodeMRLAql/view?usp=sharing

FrYakaTKoP, 2021-11-22:
🤔 sorry i don't see anything obvious do you have discord so we can interact directly? if so i'm "FrY Sennberg#4645" and on the klipper, mainsail, voron, ratrig and fluidd server. Just hit me up with a dm please, i'm happy to help but seems flimsy to do here.

Sekenenz, 2021-11-22:
No but I'll register, thanks

closed 2021-11-28 by FrYakaTKoP

cmerkle, 2023-06-01:
Was this ever resolved? I'm trying to accomplish the same thing and am having the same issue

Sekenenz, 2023-06-01:
I don't know, I gave it up at the time.