mainsail-crew/moonraker-timelapse #17: Add logo to rendered video

Sekenenz, 2021-11-21:
I've tried to add a logo to the rendered video using the extraoutputparams: ~/logo.png -filter_complex "overlay=x=10:y=10"
And obviously it's not working because I don't really know what I'm doing.

FrYakaTKoP, 2021-11-22:
@Sekenenz 
i already have it on the list of upcoming features, the hard experimental work on how to convince ffmpeg to behave is already done. I just need some time to play around with it more and then implement it to the plugin. At the moment i'm in feature freeze because of the mainsail v2.1 beta release which use up my time to help people get the plugin up and running. Also fix bugs which appear here and there. After this first big wave, i will start implementing features again.

problem you have is that extraoutputparams doesn't allow you to specify extra input files, so with momentary state of the plugin it's not achievable. 

If you really need a watermark rendered to your timelapse videos i suggest to just render it manually from the shell. I can help you getting it setup, but there are a couple good tutorials on how to build the correct ffmpeg commands to add a watermark to it. You can either render the watermark on to the existing mp4 or go the bit harder route and directly render the frames with a watermark (i suggest you look into the moonraker.log after a render and grab the shell command out of there as a starting point)

Sekenenz, 2021-11-22:
Obviously, no rush. I'm already doing it manually with ffmpeg from the terminal, but the possibility to auto render it will be a nice add.