mainsail-crew/moonraker-timelapse #18: Unparsed config section [timelapse] detected

phoenixfire425, 2021-11-27:
I am getting this error with a fresh install of fluiddpi, klipper, moonraker.

[image]

fluidd warnings found.
Moonraker has failed plugins, please check your logs, update your configuration and restart moonraker.
timelapse
Moonraker plugin configuration can be found here.

Unparsed config section [timelapse] detected.

2021-11-27 14:30:04,118 [moonraker.py:load_component()] - Unable to load component: (timelapse)
Traceback (most recent call last):
  File "/home/pi/moonraker/moonraker/moonraker.py", line 235, in load_component
    module = importlib.import_module("components." + component_name)
  File "/home/pi/moonraker-env/lib/python3.7/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1006, in _gcd_import
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 965, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'components.timelapse'
2021-11-27 14:30:04,119 [moonraker.py:add_warning()] - Unparsed config section [timelapse] detected.  This may be the result of a component that failed to load.  In the future this will result in a startup error.
2021-11-27 14:30:04,141 [moonraker.py:_initialize_component()] - Performing Component Post Init: [machine]
2021-11-27 14:30:04,198 [shell_command.py:_check_proc_success()] - Command (systemd-detect-virt) exited with return code 1

[moonraker (5).log](https://github.com/mainsail-crew/moonraker-timelapse/files/7611888/moonraker.5.log)

phoenixfire425, 2021-11-27:
> I am getting this error with a fresh install of fluiddpi, klipper, moonraker.
> 
> [image]
> 
> fluidd warnings found. Moonraker has failed plugins, please check your logs, update your configuration and restart moonraker. timelapse Moonraker plugin configuration can be found here.
> 
> Unparsed config section [timelapse] detected.
> 
> 2021-11-27 14:30:04,118 [moonraker.py:load_component()] - Unable to load component: (timelapse) Traceback (most recent call last): File "/home/pi/moonraker/moonraker/moonraker.py", line 235, in load_component module = importlib.import_module("components." + component_name) File "/home/pi/moonraker-env/lib/python3.7/importlib/**init**.py", line 127, in import_module return _bootstrap._gcd_import(name[level:], package, level) File "", line 1006, in _gcd_import File "", line 983, in _find_and_load File "", line 965, in _find_and_load_unlocked ModuleNotFoundError: No module named 'components.timelapse' 2021-11-27 14:30:04,119 [moonraker.py:add_warning()] - Unparsed config section [timelapse] detected. This may be the result of a component that failed to load. In the future this will result in a startup error. 2021-11-27 14:30:04,141 [moonraker.py:_initialize_component()] - Performing Component Post Init: [machine] 2021-11-27 14:30:04,198 [shell_command.py:_check_proc_success()] - Command (systemd-detect-virt) exited with return code 1
> 
> [moonraker (5).log](https://github.com/mainsail-crew/moonraker-timelapse/files/7611888/moonraker.5.log)

I was able to resolve.  I uninstalled Timelapse and reinstalled.  All is working now.

closed 2021-11-27 by phoenixfire425

filemakers, 2022-10-23:
how do you uninstall and reinstall time lapse please

FrYakaTKoP, 2022-10-23:
no need to uninstall it, just re run the install.sh. Warning the installer only works on the "old" path layout (not the new "printer_data" one)

if you like to uninstall it execute following commands via ssh:

````rm -rf ~/moonraker-timelapse
rm -f ~/moonraker/moonraker/components/timelapse.py
rm -f ~/klipper_config/timelapse.cfg
````

filemakers, 2022-10-23:
I’m using a speeder pad, I also don’t understand what you mean by old and new layouts, I’m grateful for your reply though :-) also the password of flsun doesn’t seem to work on my pad.

5dman, 2023-10-08:
Reinstallation didn't help..
please, help me

r14n, 2024-01-23:
> no need to uninstall it, just re run the install.sh. Warning the installer only works on the "old" path layout (not the new "printer_data" one)
> 

thanks you helped me get timelapse loading again

Dood345, 2024-05-22:
I encountered this same issue when running moonraker-timelapse installation steps. Found out that the version of debian that was requested seemed to no longer exist as of late last year. Followed the below stack exchange and it helped:
https://unix.stackexchange.com/questions/371890/debian-the-repository-does-not-have-a-release-file
ran: `sed -i s/deb.debian.org/archive.debian.org/g /etc/apt/sources.list`