mainsail-crew/moonraker-timelapse #23: please add a uninstall script

HelgeKeck, 2021-12-10:
hi, please add a uninstall script

thank you

FrYakaTKoP, 2021-12-12:
i can add a uninstall script sure, but the user would still need to remove couple things manually (moonraker.conf and printer.cfg entries, slicer config) to remove everything. Don't know if a uninstaller script which just does unlink the .py and the .cfg and remove the dummy service (reverse the installer) would make much sense. What's your opinion on that? Maybe it's better to include a guide "how to uninstall", then a uninstaller script.

HelgeKeck, 2021-12-12:
i agree, there should be a how to unisntall, exactly like the how to install, i dont see why it should be different. atm a user needs to analyze the install script if he wants to uninstall it.

FrYakaTKoP, 2021-12-12:
i will add a guide how to uninstall

HelgeKeck, 2021-12-12:
👍

dvomang, 2023-01-10:
Did you ever make the uninstall guide?

Randy1727, 2023-01-19:
Can anyone help with uninstalling timelapse.

FrYakaTKoP, 2023-01-19:
@dvomang sorry no, still need to make a guide.

@Randy1727 
To uninstall follow

Randy1727, 2023-01-19:
Ty for that info.

zellneralex, 2023-02-14:
Solved with 3ac026158b2b849456bb9bf20c7663c667fa4d84

docs will follow shortly

closed 2023-07-29 by FrYakaTKoP