mainsail-crew/moonraker-timelapse #26: bug: when using fw retract, next travel moves uses un-retract speed, which causes bit more oozing.

vajonam, 2021-12-12:
as per @zellneralex - Yes as you do fw retraction there is a travel move that does use the speed set by the unretract of the macro

creating bug to track. Attached gocde example. 

[pikachu_1gen_flowalistik_1h51m_0.2mm_243C_PETG_ENDER3BLTOUCH.gcode.zip](https://github.com/mainsail-crew/moonraker-timelapse/files/7699239/pikachu_1gen_flowalistik_1h51m_0.2mm_243C_PETG_ENDER3BLTOUCH.gcode.zip)

zellneralex, 2021-12-14:
OK a short update here:

**Root Cause:** Analyzing the send gcode shows that fw_retract (G10/G11) ) is used. Because of that there is no F parameter set at the retract/unretract in the gcode and the last set F value comes from the unretract insight the  _WAIT_TIMELAPSE_TAKE_FRAME macro.

**Solution:** Store the current F value before the park move (as already done for other values) and restore it after the unretract in _WAIT_TIMELAPSE_TAKE_FRAME

New macros are in test and show that the problem is solved with the next released version of timelapse.cfg

zellneralex, 2021-12-14:
This is the fix for it and will be merged soon https://github.com/mainsail-crew/moonraker-timelapse/commit/4d4ef1e92bf6cfa1a2dbc4fce28749113c9e355c

FrYakaTKoP, 2021-12-15:
should be fixed with #28

closed 2021-12-15 by FrYakaTKoP