mainsail-crew/moonraker-timelapse #31: // Unknown command:"_SET_TIMELAPSE_SETUP" after install on Klipper restart

DangRock, 2022-01-01:
I can't find anything in the logs to predict an error. It looks like the Timelapse macros are not recognized by Klipper. I've searched Klippy.log and Moonraker.log for issues but find none.  

Only customization I did was add this to moonraker.cfg:

```
[timelapse]
  enabled: True
  autorender: True
  constant_rate_factor: 20
  output_framerate: 15
  output_path: /home/pi/gcode_files/Timelapses/
  frame_path: /tmp/timelapse/
  ffmpeg_binary_path: /usr/bin/ffmpeg
  snapshoturl: http://fluiddpi.local/webcam/?action=snapshot
  extraoutputparams: -filter_complex "[0]trim=0:3[hold]"
```

I see this in my fluiddpi console on reboot:

```
17:18:49 
// Klipper state: Ready
17:18:50 
$ _SET_TIMELAPSE_SETUP ENABLE=True VERBOSE=True PARK_ENABLE=False PARK_POS=back_left CUSTOM_POS_X=0.0 CUSTOM_POS_Y=0.0 CUSTOM_POS_DZ=0.0 TRAVEL_SPEED=100 RETRACT_SPEED=15 EXTRUDE_SPEED=15 RETRACT_DISTANCE=1.0 EXTRUDE_DISTANCE=1.0 PARK_TIME=0.1 FW_RETRACT=False
17:18:50 
$ HYPERLAPSE ACTION=STOP
17:18:50 
// Unknown command:"_SET_TIMELAPSE_SETUP"
17:18:50 
// Unknown command:"HYPERLAPSE"
```

Any ideas?

DangRock, 2022-01-02:
Nevermind... I missed the part where I have to include the timelapse.cfg in the printer.cfg to include those macros. That seemed like part of the "Fallback Config File" section so I didn't think I needed it.  I put that in and the unavailable macros are fixed.

closed 2022-01-02 by DangRock

scuba1976, 2023-10-15:
I have the same problem. Can you say me what I need to include on printer.cfg?

martineight, 2023-10-17:
I had the same issue recently.  You need to add the following to printer.cfg

[include timelapse.cfg]

As per the instructions in [https://github.com/mainsail-crew/moonraker-timelapse/blob/main/docs/configuration.md](url)