mainsail-crew/moonraker-timelapse #32: Where does HYPERLAPSE_CYCLE go?  Not effective in moonraker.cfg...

DangRock, 2022-01-02:
Title is it... where do I put the HYPERLAPSE_CYCLE=5 setting if I want 5-second pictures when in hyperlapse mode?  I've set it in moonraker.cfg and it doesn't cause errors, but it isn't effective. The hyperlapse cycle always starts at 30 seconds. My workaround is to put this just after the nozzle temps are reached in my Cura start gcode settings:

HYPERLAPSE ACTION=STOP
HYPERLAPSE ACTION=START CYCLE=5

I'm also wanting to switch to layermacro at the first layer... so at the end of the start gcode, I put another HYPERLAPSE ACTION=STOP and then my layer processing includes the TIMELAPSE_TAKE_FRAME. That does seem to work correctly. 

One thing I've noticed is that "Before Layer 1" seems to be AFTER the adhesion is printed, but before the object layer is printed. I'm trying to find a way to leave HYPERLAPSE on through the brim or outline, then switch to layermacro.  Any ideas there?

FrYakaTKoP, 2022-01-06:
@DangRock 

If you like to change the cycle you need to change it in the settings not manually call the macro. Preferred way to change the cycle time is in the timelapse settings (when using mainsail >2.1) or in the moonraker.conf

just calling action=stop does't tell the plugin that hyperlapse is stopped (yes the hyperlapse stops but the plugin still in hyperlapse mode). The reason why takeframe is ignored in the hyperlapse mode is that a user can have layermacro in slicer setup but decide to use hyperlapse on that same gcode to without changing slicer setup and reslice without takeframe.

If your use case is to have hyperlapse on in the first layer but deactivate it for rest of the print you have two options:

1) don't set hyperlapse mode in the plugin (mode = layermacro) and just start the hyperlapse macro in your start code ``HYPERLAPSE ACTION=START CYCLE=5`` and then after the first layer or where ever use the hyperlapse action=stop

2) would be to activate hyperlapse mode and just stop it with action stop when you like it. But then you need to simulate the "called from hyperlapse" to the take frame macro so the plugin does not ignore the takeframe call ```TIMELAPSE_TAKE_FRAME HYPERLAPSE=True```

FrYakaTKoP, 2022-01-11:
closing this issue, since the Question was answered and there is no reason to leave it open. 
@DangRock  please reopen it, if there is any follow up questions.

closed 2022-01-11 by FrYakaTKoP