mainsail-crew/moonraker-timelapse #33: Camera options like brightness and contrast

LFSPille, 2022-01-10:
Where can i put any additional camera options like the above?
Because my webcam is always much too bright

FrYakaTKoP, 2022-01-11:
@LFSPille 
Timelapse just fetches the snapshot from the streamer service and has no control over the camera at all. Also this will never find it's way into timelapse because it doesn't make sense if someone just want to have more camera settings but no timelapse, but then need to install timelapse to have the extra settings. It make way more sense to add something like that to the streamer service, which does control other settings of the camera already.

There is a improved streamer service in development which have plans to make camera settings available in future, but it's still a long road to go and therefore it will take some time until this is ready. Take a look at: https://github.com/mainsail-crew/crowsnest 

I will close this issue, because not timelapse related.

closed 2022-01-11 by FrYakaTKoP

LFSPille, 2022-01-11:
Thx a lot for the answer. So i will wait for the Service 😊 Gesendet von Mail für Windows Von: FrY SennbergGesendet: Dienstag, 11. Januar 2022 21:08An: mainsail-crew/moonraker-timelapseCc: LFSPille; MentionBetreff: Re: [mainsail-crew/moonraker-timelapse] Camera options like brightness and contrast (Issue ***@***.*** just fetches the snapshot from the streamer service and has no control over the camera at all. Also this will never find it's way into timelapse because it doesn't make sense if someone just want to have more camera settings but no timelapse, but then need to install timelapse to have the extra settings. It make way more sense to add something like that to the streamer service, which does control other settings of the camera already.There is a improved streamer service in development which have plans to make camera settings available in future, but it's still a long road to go and therefore it will take some time until this is ready. Take a look at: https://github.com/mainsail-crew/crowsnestI will close this issue, because not timelapse related.