mainsail-crew/moonraker-timelapse #34: Problems with the Park Toolhead function

Bmerino16, 2022-01-16:
Hello, I just installed the timelapse function and it is working. But I have problems with the Park Toolhead option, although I have it activated it does not work. What can I do in this case?

FrYakaTKoP, 2022-01-17:
@Bmerino16 hard to guess without more information. Can you provide a moonraker.log and klippy.log? Also which os your on? Do you may have old config from the first version of the plugin in your printer.cfg?

Bmerino16, 2022-01-19:
@FrYakaTKoP Hi, sorry or the late. here you have the the moonraker.log and klippy.log

[klippy.log](https://github.com/mainsail-crew/moonraker-timelapse/files/7893245/klippy.log)

[moonraker.log](https://github.com/mainsail-crew/moonraker-timelapse/files/7893246/moonraker.log)

For the OS version, I don't know i you mean this:
[image]

This is the only thing I have in printer.cfg

[image]

[image]

And in the moonraker.conf the update manager and the [timelapse] section.

I appreciate your help!

zellneralex, 2022-01-19:
Can you do me a favor an€ run GET_TIMELAPSE_SETUP in the console and post the result

zellneralex, 2022-01-19:
And I hope you do not have that gcode macro in your printer.cfg as that would override our macro and would explain all. Your config does not show that btw

FrYakaTKoP, 2022-01-19:
@Bmerino16 remove the "TIMELAPSE_TAKE_FRAME" Macro from the printer.cfg this is the old version without the parkfeature and because it's defined after the include of timelapse.cfg it overwrites the new one in there.

Bmerino16, 2022-01-19:
@FrYakaTKoP I just followed your recommendation and removed the "TIMELAPSE TAKE FRAME" macro. Now it works perfectly.

I really appreciate your help!

FrYakaTKoP, 2022-01-19:
your welcome! 
closing the issue because it's resolved

closed 2022-01-19 by FrYakaTKoP