mainsail-crew/moonraker-timelapse #42: Park causes printer to pause

TheLion, 2022-02-18:
When using the feature to park the nozzle, it triggers a "Print paused" state, everytime a picture is taken. 

Which results in a notification when using the Moonraker Telegram (https://github.com/Raabi91/moonraker-telegram) plugin. 

Is there a way to combat this? Apart from not parking the nozzle of course ;-)

zellneralex, 2022-02-18:
The PAUSE is absolutely needed otherwise the concept will not work.
The macro provides a Flag called printer[“gcode_macro TIMELAPSE_TAKE_FRAME“].is_paused and the designer of the bot should omit the notifications if that and printer.pause_resume. is_paused Both set to true.

Sorry you need to wait until the designer of the bot has done that. I will add a feature request to the project.

zellneralex, 2022-02-20:
I was told the telegram bot is updated.

closed 2022-02-20 by zellneralex

TheLion, 2022-02-20:
Great! Will run a test when I am back home next week. 
Thanks!