mainsail-crew/moonraker-timelapse #43: _SET_TIMELAPSE_SETUP in Console at each time Moonraker restart

WSchlesner, 2022-02-21:
I am running moonraker-timelapse, following the installation instructions on the GitHub. I have gotten it to work but each time I restart Moonraker. I am getting the below command in my console each time.

""_SET_TIMELAPSE_SETUP  ENABLE=True VERBOSE=False PARK_ENABLE=False PARK_POS=back_left CUSTOM_POS_X=0.0 CUSTOM_POS_Y=0.0 CUSTOM_POS_DZ=0.0 TRAVEL_SPEED=100 RETRACT_SPEED=15 EXTRUDE_SPEED=15 RETRACT_DISTANCE=1.0 EXTRUDE_DISTANCE=1.0 PARK_TIME=0.1 FW_RETRACT=False""

""HYPERLAPSE ACTION=STOP""

Console:
[image]

Below is my config because I do not have access to change the settings in the GUI:
[image]

Any ideas on how to fix this, or have my settings override them? When I go to add more settings in the moonraker.cfg it breaks the timelapse.

FrYakaTKoP, 2022-02-21:
@WSchlesner 
Sorry this behaviour can't be changed within the plugin or config. The webui needs to suppress this messages (like mainsail does). You need to open a feature request on fluidd to get those messages hidden. I thought they started to work on a timelapse integration in fluidd, but don't know the state of it.

WSchlesner, 2022-02-22:
@FrYakaTKoP 
Thanks, I will reach out and see. Going to start to use Mainsail to see if I can get used to the GUI.

closed 2022-02-22 by WSchlesner