mainsail-crew/moonraker-timelapse #44: Update-Manager states "Invalid"

Silikowski, 2022-02-24:
When trying to install the timelapse-component, the update-manager sets the status of the timelapse to "invalid".

mainsail v2.1.2
moonraker v0.7.1-424-g501af62

its a fresh/clean installation of mainsail.

moonrakerlog shows:

`2022-02-24 10:37:42,557 [update_manager.py:initialize()] - 
Initializing Client Updater: 'mainsail',
Channel: stable
path: /home/pi/mainsail
2022-02-24 10:37:42,578 [shell_command.py:_check_proc_success()] - Command (git -C /home/pi/moonraker-timelapse status -u no) successfully finished
2022-02-24 10:37:42,595 [shell_command.py:_check_proc_success()] - Command (git -C /home/pi/moonraker-timelapse config --get branch.main.remote) successfully finished
2022-02-24 10:37:42,614 [shell_command.py:_check_proc_success()] - Command (git -C /home/pi/moonraker-timelapse remote get-url origin) successfully finished
2022-02-24 10:37:42,640 [shell_command.py:_check_proc_success()] - Command (git -C /home/pi/moonraker-timelapse fetch origin --prune --progress) exited with return code 255
2022-02-24 10:37:43,172 [shell_command.py:_check_proc_success()] - Command (git -C /home/pi/moonraker-timelapse fetch origin --prune --progress) exited with return code 255
2022-02-24 10:37:43,719 [shell_command.py:_check_proc_success()] - Command (git -C /home/pi/moonraker-timelapse fetch origin --prune --progress) exited with return code 255
2022-02-24 10:37:44,265 [shell_command.py:_check_proc_success()] - Command (git -C /home/pi/moonraker-timelapse fetch origin --prune --progress) exited with return code 255
2022-02-24 10:37:44,810 [shell_command.py:_check_proc_success()] - Command (git -C /home/pi/moonraker-timelapse fetch origin --prune --progress) exited with return code 255
2022-02-24 10:37:45,313 [git_deploy.py:initialize()] - Git Repo timelapse: Initialization failure
Traceback (most recent call last):
  File "/home/pi/moonraker/moonraker/components/update_manager/git_deploy.py", line 325, in initialize
    await self.fetch()
  File "/home/pi/moonraker/moonraker/components/update_manager/git_deploy.py", line 534, in fetch
    f"fetch {self.git_remote} --prune --progress")
  File "/home/pi/moonraker/moonraker/components/update_manager/git_deploy.py", line 778, in _run_git_cmd_async
    raise self.server.error(f"Git Command '{cmd}' failed")
utils.ServerError: Git Command 'fetch origin --prune --progress' failed
2022-02-24 10:37:45,313 [git_deploy.py:refresh()] - Error Refreshing git state
Traceback (most recent call last):
  File "/home/pi/moonraker/moonraker/components/update_manager/git_deploy.py", line 55, in refresh
    await self._update_repo_state()
  File "/home/pi/moonraker/moonraker/components/update_manager/git_deploy.py", line 61, in _update_repo_state
    await self.repo.initialize(need_fetch=need_fetch)
  File "/home/pi/moonraker/moonraker/components/update_manager/git_deploy.py", line 325, in initialize
    await self.fetch()
  File "/home/pi/moonraker/moonraker/components/update_manager/git_deploy.py", line 534, in fetch
    f"fetch {self.git_remote} --prune --progress")
  File "/home/pi/moonraker/moonraker/components/update_manager/git_deploy.py", line 778, in _run_git_cmd_async
    raise self.server.error(f"Git Command '{cmd}' failed")
utils.ServerError: Git Command 'fetch origin --prune --progress' failed`

as i could clone the repo with no problem, i guess the GIT-URL in moonraker.conf should still be valid as

`[update_manager timelapse]
type: git_repo
primary_branch: main
path: ~/moonraker-timelapse
origin: https://github.com/mainsail-crew/moonraker-timelapse.git`

FrYakaTKoP, 2022-02-24:
@Silikowski 
could you excute following commands via ssh:
``` cd ~/moonraker-timelapse
git status```
and post the output?

Silikowski, 2022-02-24:
> @Silikowski could you excute following commands via ssh:
> 
> ```
> cd ~/moonraker-timelapse
> git status 
> ```
> 
> and post the output?
Sure:

pi@3dp:~/moonraker-timelapse $ git status
On branch main
Your branch is up to date with 'origin/main'.

nothing to commit, working tree clean

FrYakaTKoP, 2022-02-24:
@Silikowski 
That's odd, sorry i can't reproduce your issue. You may just delete the directory and retry afterwards.

````
rm -rf ~/moonraker-timelapse
cd ~/
git clone https://github.com/mainsail-crew/moonraker-timelapse.git
bash ~/moonraker-timelapse/install.sh
````

then click on the "check for updates" button within mainsail to see if the issue persists. If so i need to talk to arksine since it's a moonraker/update-manager issue and not timelapse related.

Silikowski, 2022-02-25:
> @Silikowski That's odd, sorry i can't reproduce your issue. You may just delete the directory and retry afterwards.
>
> ```
> rm -rf ~/moonraker-timelapse
> cd ~/
> git clone [...]

Did this (did it already once before posting for sure) - now it works.

> timelapse
> v0.0.1-113-ga4d31b7

Thank you for your assistance!

closed 2022-02-25 by FrYakaTKoP