mainsail-crew/moonraker-timelapse #50: Missing check for `wget` on the install script

pedrolamas, 2022-04-14:
The moonraker component is using `wget` to download the snapshots yet the install script is not checking if `wget` is installed!

I get that standard Debian/Ubuntu install will most likely have `wget` installed, but that is not the case with Docker images as to keep their size small (normally one will only install the bare minimums on the image)

I would thus recommend adding a check for `wget`, just like the exiting one for `ffmpeg`

FrYakaTKoP, 2022-04-25:
Thanks for your input and sorry for late response.

I have plans to move away from wget and use the tornado httpclient in future (not 100% sure yet), but if the plugin stays with wget i will add the check to the installer.

zellneralex, 2023-02-14:
Solved with new installer 3ac026158b2b849456bb9bf20c7663c667fa4d84

closed 2023-02-14 by zellneralex