mainsail-crew/moonraker-timelapse #52: Wrong camera in Timelapse Status

sprior, 2022-05-04:
I have two cameras defines, "Upper" and "Lower".  In the Settings page for Timelapse I have the Lower camera selected, but on the Timelapse Status page it is showing the view of the Upper camera.

matmen, 2022-05-04:
I can reproduce it, and it looks like it's because fluidd saves all settings in a `fluidd` namespace (under key `cameras.cameras`, whereas moonraker-timelapse expects them to be in the `webcams` namespace.

@FrYakaTKoP As database entry namespaces/keys aren't specified by moonraker and changing this in fluidd would require migration of that setting (including a big overhaul over how fluidd saves data), do you think we could add a client-specific config location here? I'm happy to PR it if you'd like.

sprior, 2022-05-04:
Thanks all for looking at this.  Is it possible for me to add an override to a config file while waiting for the real fix?

matmen, 2022-05-04:
for now you can set the camera URL in the timelapse config, for example:
```
[timelapse]
snapshoturl: http://fluiddpi/webcam2/?action=snapshot
```

FrYakaTKoP, 2022-08-22:
noice! :)

pedrolamas, 2022-09-02:
This issue should now be fixed in the recently released Fluidd 1.20.0.

**For the devs:** in this new version we moved the `fluidd.cameras` namespace to the `webcams` namespace, following the same field names that Mainsail uses and adding some extra of our own (so there should not be any conflict and cameras should work on both web UIs)

This change also makes it compatible with the new Moonraker Webcam APIs.

FrYakaTKoP, 2022-09-05:
i will close this issue, since it should be resolved 
feel free to reopen if needed

closed 2022-09-05 by FrYakaTKoP