mainsail-crew/moonraker-timelapse #53: Park feature not working

Jonspeeth, 2022-05-11:
Hi, I just set up moonraker timelapse, and I have the park feature enabled in the config, but it's not parking when it is supposed to send the screenshot gcode or whatever. It takes a picture each layer as enabled in my slicer, but it doesn't park in the same spot. I've attatched both my printer.cfg and timelapse.cfg, if anyone knows what's up, or maybe if I'm doing something wrong, please let me know. Thanks.
[printer.txt](https://github.com/mainsail-crew/moonraker-timelapse/files/8666010/printer.txt)
[timelapse.txt](https://github.com/mainsail-crew/moonraker-timelapse/files/8666011/timelapse.txt)
.

FrYakaTKoP, 2022-05-14:
can you provide a moonraker.log and klippy.log? So i can look into it, i can't tell from the configs alone. Just a hint: you can always test the park feature by homing and heat up your printer then hit the takeframe macro.

Jonspeeth, 2022-05-14:
[klippy.log](https://github.com/mainsail-crew/moonraker-timelapse/files/8693565/klippy.log)
[moonraker.log](https://github.com/mainsail-crew/moonraker-timelapse/files/8693566/moonraker.log)
Here you go

FrYakaTKoP, 2022-05-15:
sorry nothing obvious found, only thing is you don't have the update_manager entry in your moonraker.conf. How did you install the plugin? 

can you run couple commands via ssh to and post the output here, for further investigation of the Problem?
```
ls -la ~/moonraker/moonraker/components/timelapse.py
```

```
ls -la ~/klipper_config/timelapse.cfg
```

```
cd ~/moonraker-timelapse/component
git status
```

Jonspeeth, 2022-05-15:
responses in terminal:

lrwxrwxrwx 1 pi pi 51 May  7 19:13 /home/pi/moonraker/moonraker/components/timelapse.py -> /home/pi/moonraker-timelapse/component/timelapse.py

lrwxrwxrwx 1 pi pi 56 May  7 19:13 /home/pi/klipper_config/timelapse.cfg -> /home/pi/moonraker-timelapse/klipper_macro/timelapse.cfg

On branch main
Your branch is up to date with 'origin/main'.

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)

	modified:   ../klipper_macro/timelapse.cfg

no changes added to commit (use "git add" and/or "git commit -a")

Jonspeeth, 2022-05-15:
Whenever klipper starts up, I get a message like this in the console 

_SET_TIMELAPSE_SETUP  ENABLE=False VERBOSE=False PARK_ENABLE=False PARK_POS=back_left CUSTOM_POS_X=10.0 CUSTOM_POS_Y=10.0 CUSTOM_POS_DZ=0.0 TRAVEL_SPEED=100 RETRACT_SPEED=15 EXTRUDE_SPEED=15 RETRACT_DISTANCE=1.0 EXTRUDE_DISTANCE=1.0 PARK_TIME=0.1 FW_RETRACT=False

Even though park is enabled in the config, it says PARK_ENABLE=False.

FrYakaTKoP, 2022-05-16:
since you have modified the macro file "timelapse.cfg" (what you shouldn't need to), raises two questions: What did you change in there? And how do you do set the "config"?

The correct way is to use the settings in your webinterface (mainsail/fluidd) or as a fallback you could make settings in the moonraker.conf.

Jonspeeth, 2022-05-18:
I just followed a guide, and it said to change the coords for custom position parking.

FrYakaTKoP, 2022-05-18:
Can you point me to the Guide so i can inform the writer? As said a user shouldn't change the file at all, settings (including custom positions) where changed either in the Ui or in the moonraker.log (first is preferred)

Jonspeeth, 2022-05-23:
Ok that worked, thanks a lot. As for the guide I saw it was just some youtube video 🤷‍♂️

closed 2022-05-23 by Jonspeeth