mainsail-crew/moonraker-timelapse #54: after clicking first tiime render during print all later renders are the same length without the extra layers.

harellevy, 2022-05-15:
after clicking render during print it saves to that exact layer as mp4, all the other times during print it exports the same file with different filename and without the new layers that have been captured. Still don't know what would happen at the end of the print. I'll update.

[image]
[image]

harellevy, 2022-05-15:
running on fluidd UI

FrYakaTKoP, 2022-05-15:
rendering isn't intended but should work fine. Are you sure there are still new frames captured?

harellevy, 2022-05-15:
@FrYakaTKoP Thanks for the quick reply, yes, you can see behind the rendered modal in the screenshot the last frames. I just saw it missed frames 40-44 so it skipped the rest. but after 44 it continued to capture. It's really weird.

harellevy, 2022-05-15:
[image]

harellevy, 2022-05-15:
[image]

harellevy, 2022-05-15:
it keeps printing. but layers 40-44 wasn't captured.
Now this was the last captured frame currently printing
[image]

harellevy, 2022-05-15:
This is what i get in the uncaptured frames:
```
{"error": {"code": 404, "message": "Not Found", "traceback": "Traceback (most recent call last):\n\n  File \"/home/pi/moonraker-env/lib/python3.7/site-packages/tornado/web.py\", line 1704, in _execute\n    result = await result\n\n  File \"/home/pi/moonraker/moonraker/app.py\", line 690, in get\n    self.root, absolute_path)\n\n  File \"/home/pi/moonraker-env/lib/python3.7/site-packages/tornado/web.py\", line 2778, in validate_absolute_path\n    raise HTTPError(404)\n\ntornado.web.HTTPError: HTTP 404: Not Found\n"}}
```

FrYakaTKoP, 2022-05-16:
strange that there is a 404 in some frames. The url which get captured shouldn't change randomly, therefore a "not found" shouldn't be possible after a sucessfull capture.

can you upload the full moonraker.log of this print?

closed 2022-12-18 by FrYakaTKoP