mainsail-crew/moonraker-timelapse #56: Thumbnail not loading: 401 unauthorized

emilevirus, 2022-05-28:
It might be a config issue but thumbnails and frames won't load. Any idea why?
Error:

error: {
code: 401,
message: "Unauthorized",
traceback: "Traceback (most recent call last):

  File "/home/pi/moonraker-env/lib/python3.9/site-packages/tornado/web.py", line 1681, in _execute
    result = self.prepare()

  File "/home/pi/moonraker/moonraker/app.py", line 494, in prepare
    self.current_user = auth.check_authorized(self.request)

  File "/home/pi/moonraker/moonraker/components/authorization.py", line 675, in check_authorized
    raise HTTPError(401, "Unauthorized")

tornado.web.HTTPError: HTTP 401: Unauthorized (Unauthorized)
"
}

FrYakaTKoP, 2022-06-03:
@emilevirus  sorry but the timelapse plugin only creates the thumbnails. Requesting the file is made by your Ui (mainsail or fluidd) and moonraker filemanager sends it.  I can't tell you where it is failing, the only thing this error message says is that the request has a permission problem. Either your Ui makes a bad request which isn't possible to fullfill by moonraker. Or moonraker can't read the file from your filesystem because of permission problems. Or your client wasn't authorized in the first place (unlikely)

please upload your full moonraker.log so i can investigate that and either help you to solve the problem or guide you to the right GH page to raise a issue there.

emilevirus, 2022-06-03:
[moonraker.log](https://github.com/mainsail-crew/moonraker-timelapse/files/8834950/moonraker.log)
Here

matmen, 2022-06-03:
This is a known issue with `force_logins` set to `True`. Basically, only the gcode thumbnails are whitelisted, but the timelapse thumbnails aren't, so requesting those with authentication forced results in an error.

FrYakaTKoP, 2022-10-17:
seems fixed in fluidd with the commits above, closed.

closed 2022-10-17 by FrYakaTKoP