mainsail-crew/moonraker-timelapse #58: Run alternative/custom image-capture scripts layer change?

jbguberman, 2022-06-04:
This is probably an edge use-case, but would it be at all possible to allow for customizing how snapshots are taken? I have this up and running and taking fantastic 1080p timelapses. But, my Raspberry Pi Cam HQ (and the DSLRs some people use for Octolapse) is capable of 4k stills. With Octolapse, you can provide custom scripts that dictate how images are captured. For example, rather than using ffmpeg, you might use Raspistill/libcamera/gphoto to take a 4k still shot with the ability to save as RAW for post-processing, if desired (e.g., [1](https://alfmikael.no/2021/05/07/how-to-get-a-4k-timelapse-with-raspi-hd-camera-on-octoprint/), [2](https://www.jeffgeerling.com/blog/2021/beautiful-3d-print-time-lapses-my-nikon-d700-and-octolapse)). I'm sure there's probably a way to hardcode an alternate capture method (albeit, my coding skills aren't up to the task), but it would be nice if I could point to a small shell or python script from the `[timelapse]` section of `moonraker.cfg`. So basically, (unless I supplied moonraker-timelapse with a broken script), moonraker-timelapse would work normally right up to the point of image capture in accordance with my config settings, would execute my custom script, and then would go back to business as usual until the next image capture). I think this would probably make more sense for layermacro mode, if it's possible at all.

FrYakaTKoP, 2022-06-04:
You are not the first one asking for such, the reason why i didn't included it yet is security concerns. If i implement such functionality i would need to make sure that external script isn't editable within the webinterface. Sure this would be the most userfriendly approach, but then a "attacker" could run arbitrary code on the server (maybe even with evelated privileges, because some stuff need root privileg to function). And im 100% certain that if i implement a "just point to your script" setting, people will place there scripts inside ``klipper_config`` folder or similar so they can edit it within the webui (there will even be YT video telling them to do), without recognising the potential risks involved. 

 My plan is (have other higher prios but it will come some day) to have premade control wrappers for most common stuff like Gphoto and raspistill and limit the "webexposed" settings not to run/edit code directly. Even some stuff is pretty easy to achive like a gpio control (for a trigger of a DSLR) or raspistill. Doing Gphoto is pretty hard, since i dont got any Gphoto compatible camera also not all supported cameras have same functionality/needs. As said i will look into it in future, but there are others things i need to do before that and atm im very distracted from timelapse because of real life. I will return and work more on timelapse, but atm time spent is limited.

jbguberman, 2022-06-04:
That makes sense, and I appreciate your concerns about security! Having premade control wrappers sounds like a great solution, if and when you have time for it.

Sonicboom247, 2022-07-04:
I would love to see something like this, maybe also include an option to align capture location with zseam or add to the Front Left, Front right options so they align with the models Front left most point.  currently I have to calculate where each model will be, find the furthest left coordinate and then setting that as the capture location.. for each model.

boecko, 2024-02-24:
Hi,

i've made a pull request for that
https://github.com/mainsail-crew/moonraker-timelapse/pull/154

This allows me to use my Canon EOS R6 or 70D for that job.

boecko, 2024-02-24:
Hi,

i've change my mind. I made a direct gphoto2 integration.
see https://github.com/mainsail-crew/moonraker-timelapse/pull/155

```
[timelapse]
use_gphoto2=True
```

thats' it.
The DSLR Cam has to be adjusted by hand. E.g just medium-jpegs, no raws etc.
One could integrate these things via 
```
gphoto2 --set-config /main/imgsettings/imageformat=...
```
but that's really out of the scope of this plugin.

boecko, 2024-02-25:
Made with `use_gphoto2=True` and a Canon EOS 70D
```
/main/capturesettings/focusmode=Manual
/main/imgsettings/iso=1000
/main/imgsettings/imageformat=Small Fine JPEG
````
https://github.com/mainsail-crew/moonraker-timelapse/assets/129901/1ec92200-154d-4983-8819-6abc8ccd7868