mainsail-crew/moonraker-timelapse #59: // Timelapse: disabled, kake frame ignored (Help)

leelydlml, 2022-06-24:
Hey for some odd reason my console says 
// Timelapse: disabled, take frame ignored
and I went to the timelapse.conf file and changed
[gcode_macro TIMELAPSE_TAKE_FRAME]
description: Take Timelapse shoot
variable_enable: True
variable_takingframe: True
to saying true and once I save and restart it just resets it to say false. 
I believe this is the issue for it ignoring the take frame command but I am new to coding and moonraker.
[Config Files.zip](https://github.com/mainsail-crew/moonraker-timelapse/files/8981655/Config.Files.zip)

fbeauKmi, 2022-07-04:
Your problem seems to be the moonraker.conf . Replace all the `=` by `:` in timelapse section

closed 2022-12-18 by FrYakaTKoP

JackBailey, 2023-12-05:
If anyone has this error and their conf already uses `:`, try just resetting your settings for the timelapse and configuring it again. Doesn't look like it changed anything but the timelapse works again 🤷

zemlyanskii-artem, 2025-01-08:
В настройках тайлапса нужно выключить Использовать откат из прошивки и все работает !
я вот только установил сделал однин тайлапс и все отказался печатать ! долго я икал причину и раз 10 все по кругу случайно выяснил причину!
[image]
надеюсь другим людям поможет !)