mainsail-crew/moonraker-timelapse #60: Hyperlapse still running after "Done printing file"

fbeauKmi, 2022-07-06:
module name:  component/timelapse.py

Conditions :  
- mode : layermacro
- hyperlapse started with gcode HYPERLAPSE ACTION=START

When "Done printing file" occurs the hyperlapse continue as the condition to stop it is (line 503)
```
            # stop hyperlapse if mode is set
            if self.config['mode'] == "hyperlapse":
```
Could it be replaced by 
```
            # stop hyperlapse if running
            if self.hyperlapserunning:
```
Sorry if it's not clear, I'm quite uncomfortable with in english. I found it trying HYPERLAPSE command to set CYCLE from gcode file.

FrYakaTKoP, 2022-07-20:
i will look into this, when i understand it right you never use the hyperlapse mode and start the macro by yourself/gcode? I stop the hyperlapse if the mode is change from hyperlapse to layerchange, mid print and thought i catched all cases where the loop was still running. Since you start it by yourself easiest fix on your side would be to also stop it manually, but anyway it seems fairly simple to force stop it on print end. I will change it this weekend.

fbeauKmi, 2022-07-21:
I find convenient to drive timelapse from slicer. Starting HYPERLAPSE macro is the way I found to set cycle easily. It also allow to start Timelapse at well choosed time (eg after skirt, before first layer, ...) .  Thanks for your work

closed 2022-10-27 by FrYakaTKoP