mainsail-crew/moonraker-timelapse #67: Probleme timelapse with ideamaker or ??

Egalistel, 2022-08-21:
Hello everyone, I am having some problems with the timelapse and ideamaker option.

At start gcode, I have purge line section. When printing with the timelapse function activated by layer change, this at the beginning is repeated at least 2 or even 3 times. can't find why?
moreover in the console, I have a lot of command error message, a text that tries to be displayed somewhere?

if anyone has any info?

I note you if below my gcode in idea:

`Start Gcode 
G21
G90
M82
G28
G92 E0
G1 Z15.0 F60
G0 E3 F200
G92 E0
G1 X10.1 Y20 Z0.28 F5000.0 ;Move to start position
G1 X10.1 Y160.0 Z0.28 F3000.0 E15 ;Draw the first line
G1 X10.4 Y160.0 Z0.28 F3000.0 ;Move to side a little
G1 X10.4 Y20 Z0.28 F3000.0 E30 ;Draw the second line
G92 E0`

`layer change gcode
TIMELAPSE_TAKE_FRAME`

And here is an example of what I can find in the console:
[image]

I don't know, do you need to provide more information?
I'm on mainsail updated

Anyway, thanks in advance if you can help me.

[klippy.log](https://github.com/mainsail-crew/moonraker-timelapse/files/9389313/klippy.log)
[moonraker.log](https://github.com/mainsail-crew/moonraker-timelapse/files/9389315/moonraker.log)

FrYakaTKoP, 2022-08-22:
sorry i'm not familiar with ideamaker, but it looks like there is something wrong within the Slicer setup. Klipper seems to recieve incomplete "TIMELAPSE_TAKE_FRAME" gcode lines. If this wouldn't be strange enough it looks like sometimes it's shorted to "RAME" but sometimes to "AME". Could you upload your sliced Gcode (may slice something neutral if you can't share the model, but please test it before hand to make sure it got the same behaviour).

As said i'm not familiar with ideamaker, but since Vez is using it successfully with the plugin i'm sure we will get it running for you too. How did you add the "layer change macro"? May look at the intructions here: https://github.com/mainsail-crew/moonraker-timelapse/blob/main/docs/configuration.md#ideamaker

-FrY

Egalistel, 2022-08-26:
Hello and sorry, I had not seen the reply notification.
I place here the gcode of a cube created by Ideamaker, but I encounter the same problem with Cura.
I reinstalled all of Mainsail in the meantime in case or during a false manipulation I would have touched something, but that does not change anything.

I place here a drive link where you can find the Gcode as well as a video showing you the start of printing with the loop of the start gcode, between 2 and 3 times before the launch of the print, which is not done when the timelapse function is disabled.

In terms of messages in the console, this also only comes when the timelaspe function is active, I don't have this kind of problem when it is off
thanking you for your reply and your help.

[image]

Link drive : [https://drive.google.com/drive/folders/1ECVbt6q1TkrHVRF7RAIV11ofzdgYOLq9?usp=sharin](url)
[image]
g

zellneralex, 2022-08-29:
Please add a gcode file showing that error.

Egalistel, 2022-08-29:
The klipper installer is the mainsail version made available directly by raspberry imager and simply updated once installed through the machine interface.
I have no error message in the console if I do not place the gcode TIMELAPSE_TAKE_FRAME or if I deactivate accelerated (timelaspe) in the window before launching the print
[image]
[image]
.

zellneralex, 2022-09-04:
Sorry for my late response, I still try to reproduce it. Can you tell use what PI you are using and how much rum it has?

Egalistel, 2022-09-04:
no worries about the response time :)

I'm using a Raspberry Pi Zero2W with 512MB RAM

The printing machine is a core x/y from tronxy (x5sa), the update file was made via the Pi terminal ...

Here are the latest updated versions available on klipper.

If necessary, I can completely make a complete image of the Klipper SD card if that can be useful
[image]

zellneralex, 2022-09-04:
Thanks, ok I have a PI zero 2W here that I can place in my Voron V2. That runs normally a PI4 and with that setup I can not reproduce it. But that will take some time.

No you do not need to make an image as you use a MainsailOS.

Egalistel, 2022-09-04:
it reproduces exactly the same thing with cura , the beginning of the gcode is executed several times before the start of the printing, and the same kind of error message in the console.

here is the gcode of the calidragon, made with cura and also with ideamaker.
[Cura_Cali-Dragon_v1.zip](https://github.com/mainsail-crew/moonraker-timelapse/files/9485031/Cura_Cali-Dragon_v1.zip)
[image]

zellneralex, 2022-10-02:
Sorry, I needed some time to get all setuped.

I tried it now on a delta using a pi zero 2W. And run never in the Problem you reported. I am honestly not sure how to proceed from here. Have you tried to update klipper? I use v0.10.0-591-ga2482d4f for my latest tests.

Egalistel, 2022-10-22:
Sorry, I had a problem with the motherboard and I had to wait to receive the new one which is not of the same revision.
On top of that with all the modifications to mainsail, I started completely from scratch.

I am now on the versions:
klipper
v0.10.0-610-g0d9b2cc1
and
time lapse
v0.0.1-121-ga20a1f0

I no longer have any strange messages in the console, the timelapse was active by default, but without the movement of the head, so I didn't even realize it right away since the problems were already starting on the line of purges.

So much the better for me and a pity at the same time because the cause remains totally unknown...

I don't know if it should be solved, but in any case thank you for the time taken to try to find where the problem came from.

zellneralex, 2022-10-23:
There where a fix in klipper https://github.com/Klipper3d/klipper/commit/f4440267e63dd372265fb223dc428ae3a83bf314 that might be eliminate your problem.

I tried it before and after the fix and never seen it.

Imwill close it for now until it pops up at some one else again

zellneralex, 2022-10-23:
Can not been reproduced.

Even the opener does not see it any more after new HW and a fresh install.

closed 2022-10-23 by zellneralex

ColdBeer72, 2023-02-24:
I'm sorry, but it looks like the error get back again -at least for me-:
I double checked my gcode, it's OK, but everytime there is a "TIMELAPSE_TAKE_FRAME" line, I receive the same error in console:
```
00:50:42
Unknown command:"AME"
```
Software:
Klipper: v0.11.0-116-gd8811717
Moonraker: v0.7.1-897-g1e7be45
Timelapse: v0.0.1-129-g3ac0261
Slicer: BambuLab Studio SoftFever Version 1.4.5

Before layer change G-CODE:
```
;BEFORE_LAYER_CHANGE
;[layer_z]
G92 E0
M117 Layer {layer_num+1}/[total_layer_count] : {filament_settings_id[0]}
TIMELAPSE_TAKE_FRAME
```

After layer change G-CODE:
`SET_PRINT_STATS_INFO CURRENT_LAYER={layer_num + 1}`

SUSPECTS:
I suspect there is a problem with timelapse plugin, as this just happen when I have it activated for a print through MainSail.

zellneralex, 2023-02-25:
@ColdBeer72 to clrift. Few things  here
- you print never thru Mainsail. Mainsail ask Klipper to load and print a file. After that Mainsail passenger looking what Klipper does.
- Neither Timelapse nor moonraker nor mainsail have any influence about the gcode file io of Klipper