mainsail-crew/moonraker-timelapse #69: Every so often, run out of disk space because of timelapse

user897943, 2022-10-15:
Please introduce a config option to watch disk space and abort timelapse if it goes over a threshold. I am constantly getting problems because of timelapse filling the SD card. This kinda option should be configurable on moonraker-timelapse itself rather than relying on scripts because filling the whole SD prevents processes from working and crashes Linux.

FrYakaTKoP, 2022-10-17:
i take it onto my list and see what i can do, no promise but it should be possible to come up with a simple solution

ColinGore, 2022-10-26:
Are you certain it's the fault of Timelapse?  I think it's the culprit sometimes, but then discover it's the fault of some huge log files.

user897943, 2022-10-26:
Definitwly timelapse. I checked with df and du. Once timelapses deleted,
problem resolves. Also disabled timelapse completely until a space
management solution is implemented in moonraker and it has been stable
since.