mainsail-crew/moonraker-timelapse #70: Multiple Camera support

asexauer, 2022-10-19:
Thanks for this great plugin. Is there any possibility to configure it to take timelapses with multiple cameras. I have a cam showing the bed and a nozzle cam and would like am Timelapse for each. I could not find any documentation how to do it.

FrYakaTKoP, 2022-10-23:
i have to support multiple cameras in future, at the moment timelapse is limited to one camera only sorry

erasmus74, 2023-03-07:
I'd like to request this feature as well. And if you point me to the what needs to be done, I'm happy to contribute code. 

Id guess making two timelapses wouldn't be too hard, add in an iterator for the Marco for taking a frame. What's tricky is the post processing. Would have to have a few presets like 2 individual recordings, a picture in picture recording, streaming one to obs and other to v4l, etc. Or at least a way to configure the command in some custom way. 

Either way, like I said, happy to push this along

ForceConstant, 2023-05-16:
Same here, would like multiple cameras, so I can see timelapse with my nozzle camera, and external camea.

astfgl, 2023-08-10:
Please add a vote for multiple cameras, with the additional ability to render one camera's images as a picture-in-picture of the other, for example the nozzle camera in the corner of the bed camera view.

One way to do this is to render the two timelapse videos separately and then have ffmpeg use one as a clipped and scaled overlay to the other and render the final video.
Another way is to use an image manipulation application like imagemagick to overlay each clipped and scaled image of one stream onto the other and then render the video from the resulting images.

A general solution to capture from an arbitrary number of camera streams and render multiple PIPs around the main camera's view is the logical conclusion.

FrYakaTKoP, 2023-08-10:
i see that as a highly requested feature already. The problem right now is to implement such feature i would to refactor a lot of core modules of timelapse like image gathering and rendering. ATM i'm still very busy with live and the small amount of time i can invest into timelapse goes in support, bugs and "fast to implement features". We may need to find a skilled python programmer with alot of enthusiast and spare time to tackle this or it won't happen in near future. Apologies for this bad news.

-FrY

bacinsky, 2024-02-21:
Hi everyone, I opened #151 to patch the multiple cameras support. It is a quick fix with minimal options. I made it because my setup was all ready but missing this feature, so I made some time to implement it. Use my repository if this will not get merged, if you really want it. Enjoy.