mainsail-crew/moonraker-timelapse #72: add timestamp overlay option

mknj, 2022-10-24:
this is related to #17 

It would be nice to be able to add a simple timestamp overlay to the video (and pictures?).

Optionally it would be nice to be able to add temperature,  speed, speed multiplier, flow, ... to the video

mknj, 2022-10-24:
simple version without nice formatting.
Instead ofthe for loop with  pbmtext it would be better to use some python library to create nice transparent jpegs from some text or even graphs. 

```sh
cd /tmp/timelapse
for f in frame*
do
  t=`stat --printf "%y\n" $f `
  o=${f/frame/overlay}
  pbmtext  "$t"|pnmtojpeg >$o
done
ffmpeg -i "frame%6d.jpg" -i "overlay%6d.jpg" -filter_complex "overlay"  output.mpg
```

FrYakaTKoP, 2022-10-26:
similar to #17. Problem with -filter_complex option it blocks some other used functionality like rotation, this could be fixed by prerender the text to jpgs or add the overlay in a second run on to the mp4. But i will look into this, could you please double check if this still works with a ffmpeg command generated by the plugin? Easiest way to get to a example command would be enable debug mode in moonraker and pull the command from the log and then modify it. Would help help down the line if you could confirm that it works with same codec and ffmpeg version used normally and provide a modified version of a plugin generated command. 

To get honest prediction, this will not happen before i have time to redo the command building system.