mainsail-crew/moonraker-timelapse #74: Add support for moving only X or Y before taking picture

steckdev, 2022-10-28:
My camera is position such that I only need the Y axis to be in the correct position before taking the screenshot. This will prevent additional movement on X that is not needed. 

I can adjust the "seam position" so that the X is in a consistent location on layer change. Although it may move the head around a little bit on the X axis. The print part would be the correct location.

This would increase the speed of the print and minimize jerking of the spool which I is causing my "co-extrusion" printed parts to be more inconsistent with photos.

zellneralex, 2022-11-18:
Need to be discussed in the team but in general possible.
I would add it to the park position option.

zellneralex, 2023-01-16:
See https://github.com/mainsail-crew/moonraker-timelapse/tree/park_modes. Will be merged soon

zellneralex, 2023-02-11:
Added the 2 park modes. Mainsail will allow to set them in V2.5

closed 2023-02-11 by zellneralex