mainsail-crew/moonraker-timelapse #75: Add custom gcode before and after TIMELAPSE_TAKE_FRAME

Sungray, 2022-11-15:
I want to change the LEDs whenever a frame is captured. Basically do a flash to get better consistent pictures. Is that possible?
Adding the gcode myself directly in timelapse.cfg makes Klipper see timelapse as "dirty".

Thanks

zellneralex, 2022-11-18:
Hi,

adding custom code is not possible. You have 2 possibilities:
1) change the file as you already did and live with the “dirty”
2) copy the file and rename it to e.g my_timelapse.cfg and include that instead of our file.
In that you can modify what you want but need to do that every time we release a new timelapse.cfg

That the only 2 ways I see at the moment. The other Idea would be generate a macro that do the led flashing and add that before and after the take frame in your slicer

closed 2022-11-18 by Sungray

zellneralex, 2022-11-19:
I tried to add as much „customization“ as possible by adding all the user setup variables. Stuff like your light on the other hand is nearly impossible as there are no common names for caselights and klipper macros are also not good in injecting custom code. If I find a solution I will add it we always open for improvement.