mainsail-crew/moonraker-timelapse #78: FFMPEG rendering error

hellobritt, 2022-11-28:
I can see the webcam stream in moonraker, but I can't get the images to render into a video using either the autorender or rendering from the timelapse files gui. Each of the images in the zip are not valid image files according to my OS. Running klipper/mainsail/moonraker on ubuntu server 22.04. 

`2022-11-28 18:20:21,901 [timelapse.py:saveFramesZip()] - saved frames: timelapse_360Scraper v2_31m_0.20mm_205C_PLA_ENDER3S1.gcode_20221128_1820_frames.zip
2022-11-28 18:20:21,907 [timelapse.py:render()] - start FFMPEG: /usr/bin/ffmpeg -r 30 -i '/tmp/timelapse/frame%6d.jpg' -threads 2 -g 5 -crf 23 -vcodec libx264 -pix_fmt yuv420p -an  '/tmp/timelapse/timelapse_360Scraper v2_31m_0.20mm_205C_PLA_ENDER3S1.gcode_20221128_1820.mp4' -y
2022-11-28 18:20:21,984 [timelapse.py:render()] - Rendering Video failed: /usr/bin/ffmpeg -r 30 -i '/tmp/timelapse/frame%6d.jpg' -threads 2 -g 5 -crf 23 -vcodec libx264 -pix_fmt yuv420p -an  '/tmp/timelapse/timelapse_360Scraper v2_31m_0.20mm_205C_PLA_ENDER3S1.gcode_20221128_1820.mp4' -y : Conversion failed!`

hellobritt, 2022-12-02:
After looking at the files in the timeplase zip file, I've determined that each file is not a jpg file type, but a html file type.

<pre>$ file frame000001.jpg
frame000001.jpg: HTML document, Unicode text, UTF-8 text
</pre>
Looking at the contents of the html file, I see

```
  <body id='splashpage'>
            <div>
            <img src='/admin/img/logo.svg' alt='Pi-hole logo' width='256' height='377'>
            <br>
            <p>Pi-<strong>hole</strong>: Your black hole for Internet advertisements</p>
            <a href='/admin'>Did you mean to go to the admin panel?</a>
            </div>
        </body>
```

So I guess my pihole  (which is running on the same ubuntu server as mainsail) is interfering in the operation of timelapse.

FrYakaTKoP, 2022-12-18:
i guess this issue was solved on the discord, with adding proper port to the snapshoturl in webcams setting.

closed, reopen if needed

closed 2022-12-18 by FrYakaTKoP

yourbuddydinec, 2023-08-17:
> 

Could you let me know the format to put it in please?