mainsail-crew/moonraker-timelapse #81: Keeps losing frames

dhstsw, 2022-12-05:
"Error taking timelapse frame"

Latest releases of Klipper, Moonraker, Fluidd and timelapse.

In moonraker.log i find a lot of lines like the following:

```
2022-12-05 20:16:57,143 [timelapse.py:getWebcamConfig()] - something went wrong gettingCam UUID:9d2f61bd-263e-4bbf-976b-21d4d7eaa2ce from Database. Exception: Key '9d2f61bd-263e-4bbf-976b-21d4d7eaa2ce' in namespace 'webcams' not found
2022-12-05 20:16:57,239 [shell_command.py:_check_proc_success()] - Command (wget http://localhost:8080/?action=snapshot -O /tmp/timelapse/frame000001.jpg) successfully finished
2022-12-05 20:17:11,822 [timelapse.py:getWebcamConfig()] - something went wrong gettingCam UUID:9d2f61bd-263e-4bbf-976b-21d4d7eaa2ce from Database. Exception: Key '9d2f61bd-263e-4bbf-976b-21d4d7eaa2ce' in namespace 'webcams' not found
```

Despite the error, it manages to get nearly all the frames but when it doesn't it stops saving them.

jsv93, 2022-12-07:
I'm gettting the same error with my setup. Just installed timelapse, but  I can never capture any frames. All latests updates but still doesnt work. Did you have any luck getting this sorted?

FrYakaTKoP, 2022-12-16:
strange since fluidd uses webcams namespace couple versions now. Did you add a webcam in settings of fluidd?

jasonb194, 2023-03-02:
I am also getting the same error. My Webcam is a http stream, not connected directly to the printer.

I was also getting errors with it trying to use a snapshot URL, which was not defined in the config:
```
2023-03-02 04:05:04,164 [timelapse.py:getWebcamConfig()] - something went wrong gettingCam UUID:cd71249e-bf32-45b2-9867-bafebb18e67a from Database. Exception: 'urlSnapshot'
2023-03-02 04:05:04,240 [shell_command.py:_check_proc_success()] - Command (wget http://localhost:8080/?action=snapshot -O /tmp/timelapse/printer/frame000001.jpg) exited with return code 4
2023-03-02 04:05:04,241 [timelapse.py:newframe()] - getting newframe failed: wget http://localhost:8080/?action=snapshot -O /tmp/timelapse/printer/frame000001.jpg
```

I added a working URL in the moonraker.conf, so it takes screenshots, but does not combine these into a video file. I am also still getting errors in the UI, and the same UUID errors as above:

```
2023-03-02 21:32:41,088 [timelapse.py:getWebcamConfig()] - something went wrong gettingCam UUID:cd71249e-bf32-45b2-9867-bafebb18e67a from Database. Exception: 'urlSnapshot'
2023-03-02 21:32:41,870 [shell_command.py:_check_proc_success()] - Command (wget http://192.168.40.180:8080/?action=snapshot -O /tmp/timelapse/printer/frame000056.jpg) successfully finished
```

jasonb194, 2023-03-03:
After some testing on my end, I found that setting up the camera in Fluidd as a MJPEG stream fixed the issue, Previously, the camera setting was "HTTP page" because my camera outputs the stream to a http page. Switching this to MJPEG stream with the same url works without issue for timelapse.

FrYakaTKoP, 2023-07-29:
part of this issue should be resolved with [3e1fa83](https://github.com/mainsail-crew/moonraker-timelapse/pull/125/commits/3e1fa830748e20db4458a30c0a677efcaf45b3ac)

closing this issue, please feel free to reopen if the problem persists

closed 2023-07-29 by FrYakaTKoP