mainsail-crew/moonraker-timelapse #83: Move out of range on first layer

CobreDev, 2022-12-12:
Whenever I start a print, I have to remove the first instance of `TIMELAPSE_TAKE_FRAME` or else I get this error:
`!! Move out of range: -0.013 -0.013 0.285 [355.142]`

I'm not sure what's causing it. The timelapse park position is set to Center.

zellneralex, 2022-12-16:
Move out of Range is in 99% a problem with absolute/relative (G90/G91) move settings. I am a little suprised as we insure in https://github.com/mainsail-crew/moonraker-timelapse/blob/main/klipper_macro/timelapse.cfg#L275 that we are in absolute before we execute the park move.

Please slice e.g. a small cube with your slicer and provide the generated gcode. I would also need a klippy.log so that I see all your macros.

CobreDev, 2022-12-16:
I figured it was something related to absolute/relative, but wasn't sure. 

Here's a 20mm x 20mm cube sliced in SuperSlicer. Right now I have `TIMELAPSE_TAKE_FRAME` inserting *after* a layer change, but it errors no matter if it's inserted before or after. 
https://pastes.io/0auz0yut4a

Here's my klippy.log:
https://pastes.io/nhtb1mzwfa

Let me know if you need anything else!

zellneralex, 2022-12-16:
That looks all ok, could you add 
`[gcode_macro DUMP_PRINT_AREA_LIMITS]
description: Debug: Print information about print volume and probeable area
gcode:
  {% set min = printer.toolhead.axis_minimum %}
  {% set max = printer.toolhead.axis_maximum %}
  {% set cfg = printer.configfile.settings %}
  {% set probe_offset = {'x' : cfg.probe.x_offset   if 'probe'   in cfg
                          else cfg.bltouch.x_offset if 'bltouch' in cfg
                          else 0,
                         'y' : cfg.probe.y_offset   if 'probe'   in cfg
                          else cfg.bltouch.y_offset if 'bltouch' in cfg
                          else 0} %}
  {% set probe_area = {'min' : {'x' : [min.x,(min.x-probe_offset.x)]|max,
                                'y' : [min.y,(min.y-probe_offset.y)]|max},
                       'max' : {'x' : [max.x,(max.x-probe_offset.x)]|min,
                                'y' : [max.y,(max.y-probe_offset.y)]|min}} %}
  {action_respond_info("Print Volume Limits:
                        Min X:%7.1f, Y:%7.1f, Z:%7.1f
                        Max X:%7.1f, Y:%7.1f, Z:%7.1f
                        Probe Area Limits:
                        Min X:%7.1f, Y:%7.1f
                        Max X:%7.1f, Y:%7.1f" % 
    (min.x,min.y,min.z,max.x,max.y,max.z,probe_area.min.x,probe_area.min.y,probe_area.max.x,probe_area.max.y))} `

and run it in the console and send the result.

CobreDev, 2022-12-17:
Here you go: 
https://pastes.io/7fl2prpx4y

I'm really struggling to figure out what's going on here!

zellneralex, 2022-12-17:
Please also ```GET_TIMELAPSE_SETUP```

CobreDev, 2022-12-17:
https://pastes.io/nhhxxebex9

On line 7m, what does the `dz:1` mean?

zellneralex, 2022-12-17:
da means delta z, on z we make only a z hop and in that case a z hop of 1 mm

zellneralex, 2022-12-18:
OK please add the following macro to your printer.cfg:
```
##  Use: DUMP_TO_CONSOLE MSG="QGL forced by PRINT_START"
[gcode_macro DUMP_TO_CONSOLE]
description: Helper: Action response
gcode: {action_respond_info(params.MSG)}
```

It simple sends the text to the console.

I used your gcode file and added some printouts:
[Test_Cube.zip](https://github.com/mainsail-crew/moonraker-timelapse/files/10253841/Test_Cube.zip)
Download & unzip it and then print it. You can cancle the print after you see  "After TIMELAPSE_TAKE_FRAME" in the console (first picture). Please send the console output so that I get a better understanding where the move out of range really happens.

Before you print the file I recommend that you open it and confirm that I did not do anything harmful. (You will find 6 DUMP_TO_CONSOLE comands in the first 110 lines). That is something I always recomand if someone ask you to print a pre sliced file.

CobreDev, 2022-12-18:
https://pastes.io/dbvrugpppz

Here's the output!

zellneralex, 2022-12-19:
Are you really sure that that is the output? I see that you open a other file in the middle of the print

zellneralex, 2022-12-19:
I also do not understand 

$ get_timelapse_setup

I checked the Klippy.log and also Goode file and both do not call it. So did you execute that manual?

CobreDev, 2022-12-19:
Weird, I missed that. I must've copied some old logs too or something, my bad!

Here's a new version:
https://pastes.io/bq0ttdaizx

zellneralex, 2022-12-19:
Wait I moment.

```
$ SET_GCODE_VARIABLE MACRO=TIMELAPSE_TAKE_FRAME VARIABLE=takingframe VALUE=False
``` 
Is the answer of the Timelapse plugin after the shoot is done. That means the error happens at RESUME.
I also see that you home to 0:0 and only move the head a little bit up.
We have seen rounding errors with homing in the past (where none really is sure why that happens)

Could you do one more test:
Replace your START_PRINT with the following 
```
```

CobreDev, 2022-12-19:
Ok interesting, that seems to have fixed it. Thank you for finding that!

closed 2022-12-19 by CobreDev