mainsail-crew/moonraker-timelapse #85: Timelapse not using the correct camera defined in UI

qoshh, 2022-12-19:
I have two cameras both streaming via mjpg_streamer using different ports on the same host.

No matter how many times I restart and select the correct one, it still uses one camera.

I wanted to try the snapshoturl in the moonraker.conf, but one of my cameras is mounted upside down and I don't see a rotation configuration in the documentation (only 'flipX' and 'flipY').

qoshh, 2023-01-20:
I just tried and it is still not using the correct camera.

fluidd v1.23.0
klipper v0.11.0-66-g4671cf2d
moonraker v0.7.1-806-ga154c5f
timelapse v0.0.1-124-g35e3104

occupiedsplash, 2023-04-14:
I am also having this issue

FrYakaTKoP, 2023-07-28:
closing this issue and leave the duplicate #109 open because it has more info in it. 

please read my comment on the other issue regarding this topic: https://github.com/mainsail-crew/moonraker-timelapse/issues/109#issuecomment-1656358897

feel free to comment on the other issue if you still have questions.

-FrY

closed 2023-07-28 by FrYakaTKoP