mainsail-crew/moonraker-timelapse #86: Rendering timelapse on cancel print.

MalHaak, 2022-12-20:
Hi,

I want it to try and render the timelapse on print stop. I think I can add the TIMELAPSE_RENDER gcode to my STOP_PRINT macro, but it says render is an "Internal use" Gcode macro. 

Is there a way I can do this?

Thanks.

zellneralex, 2023-01-10:
Yes, TIMELAPSE_RENDER, is the macro to use for that.

you miss understood the comment https://github.com/mainsail-crew/moonraker-timelapse/blob/main/klipper_macro/timelapse.cfg#L373

At the top of each macro I list a description of the variables defined inside. That should help people to understand it a little better. And in this case there is a variable render that is used only internal.

zellneralex, 2023-02-11:
No response from user so I assume it is solved

closed 2023-02-11 by zellneralex