mainsail-crew/moonraker-timelapse #87: Custom Install paths create installer issues

Supruh, 2022-12-30:
Custom install paths make issues with the installer, which assumes to find `klipper_config` under `/home/users`, however for some reason my klipper install is simply under `~klipper`. I have tried to link the folders as shown in a previous issue however now I am running into issues in mainsail where '/home/user/printer_data/config/timelapse.cfg' does not exist. As a new user to linux I feel given enough time I can figure something out myself however custom install paths would be a much appreciated addition

Supruh, 2022-12-30:
I was able to fix it however I still feel custom install paths may benefit future users who may need it

zellneralex, 2023-01-10:
We need to adopt the new moonraker paths (printer_data/…). The installer is still tailored for the old standard.

FrYakaTKoP, 2023-01-19:
i have plans to rework the old installer to work with the old and the new path system. Also a custom path parameter should be possible.

Supruh, 2023-01-22:
Cool! I ended up fixing it, but thanks for getting to it

closed 2023-01-22 by Supruh