mainsail-crew/moonraker-timelapse #90: Print is pausing when after I hit start due to unknown command.

dvomang, 2023-01-10:
[image]
I am getting this error when starting a print.

zellneralex, 2023-01-10:
That means you did not include the timelapse.cfg to your printer.cfg.

so add

[include timelapse.cfg] inside your printer.cfg

zellneralex, 2023-01-10:
See also https://github.com/mainsail-crew/moonraker-timelapse/blob/main/docs/configuration.md#define-the-gcode-macro

zellneralex, 2023-02-11:
No response from user so I assume it is fixed.

closed 2023-02-11 by zellneralex

Diegolgo, 2024-10-20:
is a know bug. just restart klipper. (or mcu).
it's happen once in a time (too many time betwen this error, always forgot, and, like y you, start to fin the problem etc etc).