mainsail-crew/moonraker-timelapse #92: add support to move toolhead after print

vic131313, 2023-01-21:
Hi ive just installed this and its awesome however is there a way to home the x axis after a print so the final frames taken are not obscured by the tool head. im not worried about parking during a print as id prefer to print with out the extra time.

FrYakaTKoP, 2023-01-27:
I would need to think about this further if this maybe would be a good future feature or not. One problem is that if i add it to the plugin the frontends need to implement it too and on the other hand it's already possible with some gcode trickery -> you can do this with a custom print end gcode, if you need help to setup such please let me know

vic131313, 2023-01-27:
Hi i found the custom park on the frontend after i'd sent this message and only now really had the time to look at what i could get it to do. rather than change anything major maybe on the frontend it could be input as 9999 to not move the tool head on that axis. so x set to 0 and y set to 9999 not sure if a larger printer would use 9999 as an acutal position or not.

Just looked at what i typed and realised its not quite what i was asking in the original comment but didnt delete it as this might be something to look at anyway.

Ill have a look at changing the print end macro with a G28 X will this run after the last frame is taken from the timelapse though

FrYakaTKoP, 2023-01-30:
you can always add 
```TIMELAPSE_TAKE_FRAME``` 
to your end gcode after what ever you like to do with your toolhead/printer to take a last frame for timelapse. Only restriction is if you use Hyperlapse mode in which case you would need to add
```TIMELAPSE_TAKE_FRAME HYPERLAPSE=True``` 
so the layer change macro prevention knows that you like to trigger the macro eventhough hyperlapse is active.

komandrik, 2023-03-15:
I have a problem with autorender in layermacro mode, below is my end script

```
M221 S100 ; reset flow
M900 K0 ; reset LA
M104 S0 ; turn off temperature
M140 S0 ; turn off heatbed
M107 ; turn off fan
G90 ; Absolute Positioning
G1 E-4 
G0 X-8 Y230 F12000 ; home X axis
TIMELAPSE_TAKE_FRAME
{if max_layer_z < max_print_height}G1 Z{z_offset+min(max_layer_z+30, max_print_height)}{endif} ; Move print BED down
M84 ; disable motors
TIMELAPSE_TAKE_FRAME
PAUSE
G4 P4000
RESUME
TIMELAPSE_RENDER
```
I'm trying to run the render in manual mode. It’s necessary, so that the nozzle goes away from the designated area, descending the style and then take the final frame, and start the auto-render.
Changed moonraker.conf by unchecking autostart render

```
[timelapse]
enabled = True
output_path = ~/timelapse/
frame_path = /tmp/timelapse/
ffmpeg_binary_path = /usr/bin/ffmpeg
autorender: False
```
But the render still starts immediately after the command M900 K0 ; reset LA
[image]

In WEB, I see a frame with a nozzle above the model, and the status window shows the last frame as I need it. As soon as I click visualization - a clip will be created exactly as I need it.

[image]

Tell me how to correctly start the render with a macro, so that it does not start automatically? Thanks

FrYakaTKoP, 2023-04-11:
sorry for the late response @komandrik 

> tell me how to correctly start the render with a macro

trigger the 'TIMELAPSE_RENDER' macro

> so that it does not start automatically

disable 'auto_render' either in the timelapse settings in mainsail or in moonraker.conf (you did later)

so you did everything right, but there is a bug

We're tracking done this "last frame doesn't get taken" bug for some time now. Only thing we (credits to JohnOCFII on discord) found out that if your add a `M400` after the last `TIMELAPSE_TAKE_FRAME` you won't loose the last frame even if you use autorender. Our guess is that it's related to gcode command buffer in klipper and stream delay compenstation. I need to find time to investigate this more, but it doesn't happen when just testing with my short gcodes so testing is very time consuming.

There happens something very odd and our guess right now is that the non blocking take frame macro executes the real frame collection after klipper thinks that all gcode and therefore the autorender is started. For now try to add 

```M400```after the last take frame call should atleast fix the lost frame bug for now

On the other hand there is seems a unrelated bug where disabling autorender doesn't work on your setup which is also odd and i need to investigate. We may need to move this to another issue though because it's not related to the "lost last frame bug". I will open a new issue for that after testing and confirming it's a bug i can reproduce on my dev setup.

komandrik, 2023-04-13:
@FrYakaTKoP Thanks for the reply, but it didn't help me. I changed the final script as per your recommendation

M104 S0 ; turn off temperature
M140 S0 ; turn off heatbed
M107 ; turn off fan
G90 ; Absolute Positioning
{if max_layer_z < max_print_height}G1 E-4 Z{z_offset+min(max_layer_z+30, max_print_height)}{endif} ; Move print BED down
;G1 E-4 F300
G0 X-8 Y230 F12000 ; home X axis
M84 ; disable motors
TIMELAPSE_TAKE_FRAME
M400
TIMELAPSE_RENDER
M220 S100 ; reset speed
M221 S100 ; reset flow
CLEAR_PAUSE ; reset pause
M900 K0 ; reset PA

Time-lapse configuration:
[timelapse]
enabled = True
output_path = ~/timelapse/
frame_path = /tmp/timelapse/
ffmpeg_binary_path = /usr/bin/ffmpeg
autorender: False
duplicatelastframe: 30

As you can see, the last frame should be shot when the table is lowered and the head is parked at the specified coordinates. After that, I added M400 and only then should the rendering take place.
But still, the last shot in the time-lapse is with the head above the detail.
[image]

https://user-images.githubusercontent.com/40995038/231857027-53ae1924-9854-4a1e-9f6d-6224cddbb9c3.mp4

If I click on the render on the right, I'll get a great time-lapse with the desired result.
[image]

https://user-images.githubusercontent.com/40995038/231857758-ceac8003-92eb-4ff5-a047-ed3a2c84bd20.mp4

It turns out that the authoring code launched by the TIMELAPSE_RENDER code does not save the last frame during the execution of the G-code. And when you run the renderer manually, after G-code finishes, it saves the video as intended.

If you need any additional experiments or additional information, I'm ready to help. P.S. I really don't want to mess around with installing and configuring a Telegram bot, having such a great plugin.

TomW1605, 2023-09-02:
hi, i am running into the same issue. was any progress made?