mainsail-crew/moonraker-timelapse #94: Issue with TIMELAPSE_TAKE_FRAME

031devstudios, 2023-02-03:
Hi guys,

I have an issue with the TIMELAPSE_TAKE_FRAME. I can see my webcam and have followed the installation according to the github page. I am having the same issue as the post below. The post says:

`edit:
We could track down WickedInvi's issue on the klipper discord: His mjpg streamer required authentication for the snapshot, because he had reinstalled the streamer with a guide which added authentication.`

Please can someone help me as to how to authenticate the mjpg streamer?

https://github.com/mainsail-crew/moonraker-timelapse/issues/15

031devstudios, 2023-02-03:
Turns out I had the incorrect snapshotURL entered. I am used the ESP32-CAM board and the correct url is http://192.168.x.xxx/capture?_cb=$

zellneralex, 2023-02-11:
User fixed it

closed 2023-02-11 by zellneralex