mainsail-crew/moonraker-timelapse #97: Any ETA on the new install script?

PigeonFX, 2023-02-11:
Just wondering if there is ETA on getting an install script that works with the old and new Klipper folder structure. Seen a free people post pull request for changes to make it compatible with the new folder structure, but all have been rejected for not being compatible with the older folder structure too.

zellneralex, 2023-02-11:
As always it is done when it is done. We doing it in our free time as service for the community. 
All the PRs up to now would break the older installs and I respect @FrYakaTKoP opinion that he wants a solution that works in both cases.

PigeonFX, 2023-02-12:
OK, just wanted to know if someone was working on it. I have put in a pull request so the install.sh that will work with old and new Klipper installs.

zellneralex, 2023-02-12:
Yes we working on it, but try to solve multiple problems 
1) old and new folder structure 
2) issue about multi instance (also old and new) 
3) issue about missing whet
4) issue about missing uninstaller

zellneralex, 2023-02-12:
I looked at your PR a minute ago, and as people do not need to use printer_data (they can name it e.g. ender_data) it would also solve the problem particularly.

PigeonFX, 2023-02-12:
I'm going to assume you mean partially rather than particularly in which case I would argue that having a partially working install script that works for 90% of people who need a script to help is better than a non-working script delayed for edge case users who have multiple installs of old and new Klipper. 

Totally get you to want to solve all the issues above, but doesn't change that the current script doesn't work for anyone but 6-month-old installs they have not updated, at least my pull request adds the original functionality of Moonraker Timelapse to both old and new installs.  

Hope that doesn't read as combative.

zellneralex, 2023-02-14:
Soon … See https://github.com/mainsail-crew/moonraker-timelapse/pull/99

zellneralex, 2023-02-14:
New installer merged 3ac026158b2b849456bb9bf20c7663c667fa4d84

closed 2023-02-14 by zellneralex