mainsail-crew/moonraker-timelapse #100: Timelapse rendering is flipped upside down

mlauber, 2023-02-15:
My timelapse renderings are upside down, but my camera image is correct when viewing it in Mainsail.  I can flip my camera in Mainsail, but that image is correct, only the timelapse renderings are upside down.  It didn't always use to be this way - I'm not sure what happened.  

Any help on fixing this would be appreciated.

FrYakaTKoP, 2023-03-03:
can you provide a moonraker.log?

mlauber, 2023-03-04:
I got it fixed by implementing flip_y: TRUE.

closed 2023-03-04 by mlauber

xdased14x, 2023-11-26:
> I got it fixed by implementing flip_y: TRUE.
> 
> Thank you though.

where did you do this at I'm having the same issues

mlauber, 2023-11-26:
Drop it in your camera configuration in Crowsnest.cfg

xdased14x, 2023-11-26:
> Drop it in your camera configuration in Crowsnest.cfg
> […](#)

thanks, ill give it a go

xdased14x, 2023-11-26:
didn't work or i put in the wrong spot

[cam 1]
mode: mjpg                              # mjpg/rtsp
port: 8080                              # Port
device: /dev/video0                     # See Log for available ...
resolution: 640x480                     # widthxheight format
max_fps: 15                             # If Hardware Supports this it will be forced, ohterwise ignored/coerced.
#custom_flags:                          # You can run the Stream Services with custom flags.
#v4l2ctl:                               # Add v4l2-ctl parameters to setup your camera, see Log what your cam is capable of.
flip_y: TRUE

mlauber, 2023-11-26:
Found it this time.  It's in Moonraker

# moonraker.conf
# [timelapse] section

#flip_x: False
#flip_y: False

xdased14x, 2023-11-26:
> Found it this time. It's in Moonraker
> 
> # moonraker.conf
> # [timelapse] section
> #flip_x: False #flip_y: False

thanks

mlauber, 2023-11-27:
No problem.  Change one of them to True and you should be set.

Matt