mainsail-crew/moonraker-timelapse #101: Done a basic video guide on new script.

PigeonFX, 2023-02-20:
knocked together a short basics guide video on getting up and running with the new script, and uninstalling. 

https://youtu.be/TyCTyk-Bq7k

FrYakaTKoP, 2023-03-03:
congrats it's one of the best videos for installing and setting up timelapse! I will consider adding a link to the docs.

On the otherhand opening a GH issue to promote your video isn't the finest way to do it. I may need to add better "how to get in touch" notes though.

closed 2023-03-03 by FrYakaTKoP

PigeonFX, 2023-03-04:
Sorry, I did look to see if this Git was using Discussions, or if there was any contact info on your profile itself, so I could post it in a more appropriate way. 

Sorry if I made extra work for you guys.