mainsail-crew/moonraker-timelapse #102: HTTPS --no-certificate-check

zu3st-de, 2023-03-01:
Hey, 
maybe we can add --no-certificate-check to the wget-command on line 455 in timelapse.py.
som it is possible to us an external IP-Camera with https. 
In my case the cam does not provide http.

FrYakaTKoP, 2023-03-03:
i like to move away from wget in future, but this sounds like a small change without much impact for those which use it without ssl. I will do couple tests tomorrow and if i don't find any show stoppers i will push that change this weekend.

FrYakaTKoP, 2023-03-12:
@zu3st-de 
just pushed 9ba308fda585ef2725cbe2f35cf28502840a43be (and 6ff8a382d1ed4d4188c18b618aef9bd3d1073a6b adds a small fix) which adds a new setting to skip cert check on frame collection.

Since there are stray versions of wget around which are compiled without SSL and it's a very niche problem, i decided not add the option as default and rather add a new setting which add the "--no-certificate-check" to the wget command.

just update to the new version and add ````wget_skip_cert_check: True
``` to your moonraker.conf and restart moonraker. After that wget will ignore wrong certificates on https servers.

Closing this issue, but feel free to reopen if it doesn't fix your main problem

closed 2023-03-12 by FrYakaTKoP