mainsail-crew/moonraker-timelapse #104: RETRACT_DISTANCE=-7.0 must be specified as positiv number

aedocw, 2023-04-04:
This is probably not an actual issue with moonraker-timelapse, much more likely it's something in my config, but I'm completely stumped trying to figure out where that -7 is coming from.

I get this error on startup:
Error evaluating 'gcode_macro _SET_TIMELAPSE_SETUP:gcode': gcode.CommandError: RETRACT_DISTANCE=-7.0 must be specified as positiv number

I don't see anything in the config files:
$ grep -i RETRACT_DISTANCE klipper_config/*
klipper_config/timelapse.cfg:# [RETRACT_DISTANCE=value] [FW_RETRACT=value] # klipper_config/timelapse.cfg: {% if params.RETRACT_DISTANCE %} klipper_config/timelapse.cfg: {% if params.RETRACT_DISTANCE|float >= 0.0 %}
klipper_config/timelapse.cfg: {% set _dummy = tl.extruder.update({'retract':params.RETRACT_DISTANCE|float|round(3)}) %} klipper_config/timelapse.cfg: {action_raise_error("RETRACT_DISTANCE=%s must be specified as positiv number" % params.RETRACT_DISTANCE)}

Where else could this be coming from? And do please point me in a better direction if there's somewhere else I should be looking for help. Really appreciate what you've built, and appreciate all your time and effort!

aedocw, 2023-04-06:
This is with the following component versions:
klipper v0.11.0-159-gfec7ddd8
crowsnest v3.0.7-3-g20ed6a8
moonraker v0.8.0-32-g31e589a
sonar v0.1.2-0-g7962a12
timelapse v0.0.1-132-gf9198e0
mainsail v2.5.0

zellneralex, 2023-04-11:
A fallback config in the moonraker.conf would overrule any client setting and would look like. https://github.com/mainsail-crew/moonraker-timelapse/blob/main/docs/configuration.md#fallback-configfile-based-moonrakerconf. So you have somewhere specified the -7 and my macro throws an error as I do not like negative numbers for that 😂

closed 2023-04-13 by aedocw