mainsail-crew/moonraker-timelapse #106: timelapse doesn't work

borislavstefanov, 2023-04-11:
where am I going wrong with the installation, after printing it only saves a rar file which is 1 kbyte with 1 picture you can't even load.  The printer is Voron 2.4 with raspbery 3+, raspbery cam ,

FrYakaTKoP, 2023-04-11:
can you provide a moonraker.log with a failed print/timelapse? My guess is that you didn't specify a correct snapshot url. If you define a snapshoturl with a stream rather than a still image, timelapse will try to get the stream and aborts collection after 2 sec (timeout), but since the stream doesn't end (mpjeg streams are continuous) the collected frame will be 0 bytes and therefore the render fails and the "save frames" zip file will only contain a empty jpeg file.

borislavstefanov, 2023-04-11:
https://github.com/borislavstefanov/Voron-2.4-delluks-files-/blob/main/moonraker%20(1).log

FrYakaTKoP, 2023-04-11:
🤔 my guess is that wget (timelapse uses internally to collect the snaphot) can't resolve `http://192.168.88.37:8080/?action=snapshot` because of the explicit ip address in there. 

the default config for `URL snapshot` in the webcams settings would be: 
```/webcam/?action=snapshot``` 
without any ip. Could you try that first, it should resolve the issue with timelapse with that.

If you need to have explicit ip in there for mainsail to work we will figure out something that will work for both timelapse and mainsail, but first we need to know if we are on the right track.

borislavstefanov, 2023-04-11:
Is that how it should be?

[timelapse]
enabled: True
mode: layermacro
snapshoturl: /webcam/?action=snapshot
gcode_verbose: True
parkhead: False
parkpos: back_left
park_custom_pos_x: 0.0
park_custom_pos_y: 0.0
park_custom_pos_dz: 0.0
park_travel_speed: 100
park_retract_speed: 15
park_extrude_speed: 15
park_retract_distance: 1.0
park_extrude_distance: 1.0
hyperlapse_cycle: 30
autorender: True
constant_rate_factor: 23
output_framerate: 30
pixelformat: yuv420p
time_format_code: %Y%m%d_%H%M
extraoutputparams: 
variable_fps: False
targetlength: 10
variable_fps_min: 5
variable_fps_max: 60
flip_x: False
flip_y: False
duplicatelastframe: 0
previewimage: True
saveframes: True

FrYakaTKoP, 2023-04-11:
yes and no. You shouldn't use moonraker.conf anymore to setup timelapse. This settings should be all in your webinterface of joice (like mainsail), but yes the the snapshoturl looks correct

FrYakaTKoP, 2023-04-11:
in mainsail (i think in fluidd too but not sure) there are cog wheels in the right top corner:

[image]

then navigate to "webcams" and setup your camera there:

[image]

[image]

then navigate to "timelapse" and add your settings there:

[image]
(make sure you select the camera you just setup or it won't work)

after that delete everything (maybe download a backup first) in the timelapse section of moonraker.conf, but leave `[timelapse]` it should look like this afterwards:

```
[timelapse]
```

now you should be able to edit settings directly in mainsail, also you can select if you like to activate timelapse when starting a print (if you may decide this print doesn't need a TL)

FrYakaTKoP, 2023-04-11:
so you can't select a camera in timelapse settings?

borislavstefanov, 2023-04-11:
yes is not active 
[image]

[image]

FrYakaTKoP, 2023-04-11:
can you show a screenshot of the webcams settings?

FrYakaTKoP, 2023-04-11:
strange after clicking "update webcam" you should be able to select the camera in timelapse settings

borislavstefanov, 2023-04-11:
I added the camera again and it started, maybe it needed some time to refresh

FrYakaTKoP, 2023-04-11:
so can you now confirm that frame collection works? Do following:
execute the "TIMELAPSE_TAKE_FRAME" macro in the dashboard or via gcode console
you now should see that frame in the status panel of the timelapse tab

borislavstefanov, 2023-04-11:
confirm works, thank you very much for your help

FrYakaTKoP, 2023-04-11:
you're welcome. Now do a print and confirm you receive a timelapse afterwards, if so close this issue.

borislavstefanov, 2023-04-11:
it works, everything is as it should be now. thank you very much again

closed 2023-04-11 by borislavstefanov