mainsail-crew/moonraker-timelapse #108: Rendering  after print problem

borislavstefanov, 2023-04-12:
hello, after printing I try to set it to rendering video, but it gives me an error. I am attaching a picture of the error it gives when I press the button

[image]

maruhe, 2023-04-14:
i get a similar error:
2023-04-14 14:13:14,742 [timelapse.py:render()] - Rendering Video failed: /usr/bin/ffmpeg -r 30 -i '/tmp/timelapse/frame%6d.jpg' -threads 2 -g 5 -crf 23 -vcodec libx264 -pix_fmt yuv420p -an  '/tmp/timelapse/timelapse_<filename>.mp4' -y : /usr/bin/ffmpeg: error while loading shared libraries: /lib/arm-linux-gnueabihf/libflite_usenglish.so.1: invalid ELF header
/lib/arm-linux-gnueabihf/libflite_usenglish.so.1 does not exist while libflite is installed

borislavstefanov, 2023-04-19:
Mainsail,  klipper on voron 2.4

maruhe, 2023-04-19:
> what OS/image are you using?
plain Raspberry OS 32bit and KIAUH
used moonraker-telegram-bot before for timelapse

borislavstefanov, 2023-04-20:
ordinary Raspberry OS 32 bit, yes I used it before but I had no problem now
after the update something happened and only

borislavstefanov, 2023-04-23:
any idea

borislavstefanov, 2023-04-29:
> > каква ОС/образ използваш?
> 
> обикновена Raspberry OS 32bit и KIAUH използваха moonraker-telegram-bot преди за timelapse

ordinary Raspberry OS 32 bit, yes I used it before but I had no problem now
after the update something happened and only

yourbuddydinec, 2023-05-01:
I noticed it did the same thing to me on a long 24 hour print which is very angering. This was important. It seemed to not save anything but the last 91 frames of the timelapse which is why its erroring...it doesn't have the first 400 frames to render,

borislavstefanov, 2023-05-01:
for me, no matter how many frames it is, it still does it, but I don't know how to fix this problem :(

yourbuddydinec, 2023-05-02:
I was able to fix it (i think, i was able to get one good timelapse so far) by changing the directory to ~/timelapse_frames in moonraker.conf and making the folder on the computer. I dont know if itll auto clear the frames though since its not the tmp folder.
```
[timelapse]
##   Following basic configuration is default to most images and don't need
##   to be changed in most scenarios. Only uncomment and change it if your
##   Image differ from standart installations. In most common scenarios 
##   a User only need [timelapse] in there configuration.
output_path: ~/timelapse/             ##   Directory where the generated video will be saved
frame_path: ~/timelapse_frames/       ##   Directory where the temporary frames are saved
ffmpeg_binary_path: /usr/bin/ffmpeg   ##   Directory where ffmpeg is installed

```

borislavstefanov, 2023-05-07:
somehow by myself today it's already fixed and records timelapse and renders after finishing printing, I just updated klipper and mainsail and now everything works, apparently something was wrong

closed 2023-05-07 by borislavstefanov

maruhe, 2023-05-07:
after all I still get 
pi@v1pi:~ $ ffmpeg -version
ffmpeg: error while loading shared libraries: /lib/arm-linux-gnueabihf/libflite_usenglish.so.1: invalid ELF header

I accept it's not a bug for here and will reinstall OS