mainsail-crew/moonraker-timelapse #109: Timelapse using the wrong camera no matter the settings.

occupiedsplash, 2023-04-14:
I'm using the fluid interface with crowsnest and 2 webcams connected and configured. It will only use the one same camera for timelapse even if the settings are set to the other camera.

It seems like the it is only grabbing port 8080 for the timelapse, because when I change the the crowsnest config to switch the ports around it will grab whatever is connected to 8080. I don't know if its an issue with how the webcamconfig works in the .py file, or if crowsnest had an update that broke the camera selection setting. 

When I want to change the camera timelapse is using, for now I just edit the crowsnest config everytime which is a pain.

Heres my software versions:
crowsnest V3.0.7
fluidd v1.23.4
klipper v0.11.0
moonraker v0.8.0
timelapse v0.0.1

FrYakaTKoP, 2023-04-19:
there seems to be a major mismatch of variable names between timelapse/mainsail/fluidd i need to look in closer this weekend.

just to make sure you are not using any snapshoturl config in moonraker.conf? Because this would always win over any config made in any Ui

occupiedsplash, 2023-04-20:
I am not, the only configs I see mention a snapshot url are in crowsnest and octoeverywhere. I have just installed octoeverywhere the other day, so it shouldnt be the root of the problem

lettore, 2023-06-10:
Same for me, it's always using the camera on /webcam endpoint no matter what you choose.

crilleb, 2023-07-16:
Same here. 

According to the Moonraker log file the getWebcamConfig method can't get the webcamconfig from the database and therefore it uses the default snapshot url (http://localhost:8080/?action=snapshot):

`[timelapse.py:getWebcamConfig()] - something went wrong gettingCam UUID:config-MoonrakerOneVision from Database. Exception: Key 'config-MoonrakerOneVision' in namespace 'webcams' not found
2023-07-16 08:55:07,859 [shell_command.py:_check_proc_success()] - Command (wget http://localhost:8080/?action=snapshot -O /tmp/timelapse/frame000001.jpg) exited with return code 4`
2023-07-16 08:55:07,805 [timelapse.py:getWebcamConfig()] - something went wrong gettingCam UUID:config-MoonrakerOneVision from Database. Exception: Key 'config-MoonrakerOneVision' in namespace 'webcams' not found
2023-07-16 08:55:07,859 [shell_command.py:_check_proc_success()] - Command (wget http://localhost:8080/?action=snapshot -O /tmp/timelapse/frame000001.jpg) exited with return code 4

FrYakaTKoP, 2023-07-28:
@crilleb sorry for the late response, didn't find much time for timelapse lately.

your 100% correct timelapse can't find the snapshot url in the webcam config, because it doesn't exist when the webcam is added using fluidd. In this case Timelapse will fallback to a hardcoded "http://localhost:8080/?action=snapshot".

I wasn't aware of this issue since i don't use fluidd myself and i don't think fluidd devs where either. I just contact one of them and it will be fixed in future for sure.

In the meantime you can add your desired snapshoturl to moonraker.conf, but be aware that this will always win over any setting done in the ui and you will need to remove this after fluidd has added this feature.

go to your moonraker.conf and search for the `[timelapse]` section and add following line to it (and then edit your url ofc)
```
snapshoturl: http://localhost:8080/?action=snapshot
```

i will leave this issue open for a while, until fluidd has added the snapshot url to webcam settings

-FrY

ToXIc69, 2024-01-24:
im having the same issue with mainsail looked for the log event as shown above and found this 

> 2024-01-24 00:32:31,072 [timelapse.py:getWebcamConfig()] - something went wrong gettingCam UUID:c270 from Database. Exception: Key 'c270' in namespace 'webcams' not found
> 2024-01-24 00:32:31,133 [shell_command.py:_check_proc_success()] - Command (wget http://localhost:8080/?action=snapshot -O /tmp/timelapse/frame000001.jpg) successfully finished

my two cams are configured but when timelapse is on it will timelapse the 1st cam regardless what i chose in the dropdown cam option. 
in crowsnest im using the device ID for the device field and if i check the snapshot id it works fine. but in the logs is just looking for the default snapshot url. how can i change it so the dropdown menu option works so i can switch between the two?

BETLOG, 2024-02-01:
Same problem.
Initially everything worked as expected with 3 completely different USB cameras connected. But now with only 2 connected timelapses keep rendering from the wrong camera, despite explicitly specifying the available and desired cams.

Can confirm that FrYakaTKoP's moonraker.conf workaround works perfectly:

```
[timelapse]
#https://github.com/mainsail-crew/moonraker-timelapse/issues/109
snapshoturl: http://localhost:8080/?action=snapshot
#snapshoturl: http://localhost:8081/?action=snapshot
#snapshoturl: http://localhost:8082/?action=snapshot
#snapshoturl: http://localhost:8083/?action=snapshot
```
But the fact that it ignores/doesnt update http://\<printer\>/#/settings#timelapse isn't ideal.

Subcode, 2025-05-18:
This still seems to be an issue in 2025. Will this get fixed?
Issue has been open since 2023..