mainsail-crew/moonraker-timelapse #116: Only take one photo during all the print.

cylonr, 2023-06-18:
Hi,

Thanks for your excellent work. 
During the print, the timelapse only took one photo and, even i turn on the Verbose G-Code, nothing appears on the console :/
After the print, the MP4 file are there but only with "one photo"

Any help.

Thanks.
[image]

ppisl, 2023-07-07:
Do you have moonraker.log? If yes, could you attach it?

zellneralex, 2023-07-17:
Please provide a small sample sliced file. Most of the times that is an error in preparing the slicer.

futnaw, 2023-12-16:
Same problem for me. Is there any solution?

zellneralex, 2023-12-16:
We did not work on it, as the user never provided any additional info.
So please provide a small sliced gcode and a moonraker.log

futnaw, 2023-12-16:
Hi,
[timelapse.zip](https://github.com/mainsail-crew/moonraker-timelapse/files/13693719/timelapse.zip)

[server]
host: 0.0.0.0
port: 7125
# The maximum size allowed for a file upload (in MiB).  Default 1024 MiB
max_upload_size: 1024
# Path to klippy Unix Domain Socket
klippy_uds_address: ~/printer_data/comms/klippy.sock

[file_manager]
# post processing for object cancel. Not recommended for low resource SBCs such as a Pi Zero. Default False
enable_object_processing: False

[authorization]
cors_domains:
    https://my.mainsail.xyz
    http://my.mainsail.xyz
    http://*.local
    http://*.lan
trusted_clients:
    10.0.0.0/8
    127.0.0.0/8
    169.254.0.0/16
    172.16.0.0/12
    192.168.0.0/16
    FE80::/10
    ::1/128

# enables partial support of Octoprint API
[octoprint_compat]

# enables moonraker to track and store print history.
[history]

# this enables moonraker announcements for mainsail
[announcements]
subscriptions:
    mainsail

# this enables moonraker's update manager
[update_manager]
refresh_interval: 168
enable_auto_refresh: True

[update_manager mainsail]
type: web
channel: stable
repo: mainsail-crew/mainsail
path: ~/mainsail

[update_manager mainsail-config]
type: git_repo
primary_branch: master
path: ~/mainsail-config
origin: https://github.com/mainsail-crew/mainsail-config.git
managed_services: klipper

### moonraker-timelapse
### Don't forget to include timelapse.cfg to your printer.cfg
### Uncomment to enable moonraker-timelapse

[update_manager timelapse]
type: git_repo
primary_branch: main
path: ~/moonraker-timelapse
origin: https://github.com/mainsail-crew/moonraker-timelapse.git
managed_services: klipper moonraker

[timelapse]
##   Following basic configuration is default to most images and don't need
##   to be changed in most scenarios. Only uncomment and change it if your
##   Image differ from standart installations. In most common scenarios 
##   a User only need [timelapse] in their configuration.
## output_path: ~/timelapse/
##   Directory where the generated video will be saved
## frame_path: /tmp/timelapse/
##   Directory where the temporary frames are saved
## fmpeg_binary_path: /usr/bin/ffmpeg
##   Directory where ffmpeg is installed

# Crowsnest update_manager entry
[update_manager crowsnest]
type: git_repo
path: ~/crowsnest
origin: https://github.com/mainsail-crew/crowsnest.git
managed_services: crowsnest
install_script: tools/pkglist.sh

# Sonar update_manager entry
[update_manager sonar]
type: git_repo
path: ~/sonar
origin: https://github.com/mainsail-crew/sonar.git
primary_branch: main
managed_services: sonar
install_script: tools/install.sh

[update_manager fluidd-config]
type: git_repo
primary_branch: master
path: ~/fluidd-config
origin: https://github.com/fluidd-core/fluidd-config.git
managed_services: klipper

[update_manager fluidd]
type: web
channel: stable
repo: fluidd-core/fluidd
path: ~/fluidd

zellneralex, 2023-12-16:
moonraker.log not conf and best case after you did a print. I will look at the gcode tomorrow