mainsail-crew/moonraker-timelapse #118: Moonraker v0.8.0-86 not apply section [timelapse]

bvitl, 2023-07-02:
After update moonraker to v0.8.0-86 it stopped running

FrYakaTKoP, 2023-07-28:
could you provide a moonraker.log? I just verified that timelapse still works with upstream moonraker (v0.8.0-125-g937c766d) right now.

valzzu, 2023-08-06:
Similar issue here but on v0.8.0-138-gfe120952 and this is what I see in the log
https://cdn.discordapp.com/attachments/1107957610558263347/1137704210583015454/image.png

FrYakaTKoP, 2023-08-10:
this error looks like something is wrong with your installation. Either you never run the installation or hard reseted the moonraker installation. Timelapse needs to copy/link it's own code in the moonraker folder and hard reseting that will delete that symlink.

The fix is pretty easy just follow the installation guide:  https://github.com/mainsail-crew/moonraker-timelapse/blob/main/docs/installation.md#installing-the-component

valzzu, 2023-08-13:
Thx, ill do that.